inthehand/32feet #2: .NET 2.0 Version?

narfanator, 2017-04-03:
Hi - I'm looking at using this for a Unity project, but Unity is (still!) limited to .NET 2.0

Should this project work; or; is there a version kicking around that should still work with the older version of .NET?

narfanator, 2017-04-03:
Actually, for that matter, is there a place I can find the DLLs to *use* 32feet? I don't think using Visual Studio's package management is compatible with Unity.

PS - I see this is a busy time for you guys, moving from CodePlex to GH!

PPS - Welcome to GitHub! I may be able to answer some questions...

peterfoot, 2017-04-03:
From version 3.5 the releases were compiled for .NET 3.5. However the source code has the appropriate ifdefs to still build against .NET 2.0. Can you describe a bit more about your scenario - what version of Unity do you need support for and what platform/OS to run on?

narfanator, 2017-04-04:
Windows 10, Unity 5.4.

peterfoot, 2017-04-04:
Thanks, looking into this further there are a couple of instances of Linq being used in the lib which is not available in .NET 2.0. This can probably be worked around with compatible code.

narfanator, 2017-04-04:
As in, Linq features that are not available? I use Linq all the time in Unity.

peterfoot, 2017-04-04:
It uses references to System.Core. I've seen workarounds suggesting referencing the 3.5 version but not sure yet if that'll work.

narfanator, 2017-04-05:
Neat. Any pointers for making this happen? I have actually never built a C# project outside of Unity.

narfanator, 2017-04-05:
Okay, think I figured out how to build it? I opened the top-level solution and built "InTheHand.Net.Personal.FX2". This built successfully into a DLL, zero issues. However, a "find in files" on #ifdef gave me nothing that looked promising (and only two results - both `ifdef  _WIN32_WCE`).

However, I downloaded and extracted the NuGet package (apparently it's just a zip?), found and extracted (just) the DLL, and stuck it Unity. It seems like it _might_ work...? Project builds with a `BluetoothClient client = new InTheHand.Net.Sockets.BluetoothClient();` line, but then errors when the application is run with:

```
SocketException: An address incompatible with the requested protocol was used.

System.Net.Sockets.Socket..ctor (AddressFamily family, SocketType type, ProtocolType proto)
InTheHand.Net.Bluetooth.Msft.SocketBluetoothClient.CreateSocket ()
InTheHand.Net.Bluetooth.Msft.SocketBluetoothClient..ctor (InTheHand.Net.Bluetooth.Factory.BluetoothFactory fcty)
Rethrow as PlatformNotSupportedException: 32feet.NET does not support the Bluetooth stack on this device.
InTheHand.Net.Bluetooth.Msft.SocketBluetoothClient..ctor (InTheHand.Net.Bluetooth.Factory.BluetoothFactory fcty)
InTheHand.Net.Bluetooth.SocketsBluetoothFactory.GetBluetoothClient ()
InTheHand.Net.Bluetooth.Factory.BluetoothFactory.DoGetBluetoothClient ()
InTheHand.Net.Sockets.BluetoothClient..ctor (InTheHand.Net.Bluetooth.Factory.BluetoothFactory factory)
InTheHand.Net.Sockets.BluetoothClient..ctor ()
ScentCollarController.Start () (at Assets/ScentCollarController.cs:216)
```

I'm poking around regarding this.

narfanator, 2017-04-05:
It's looking like it's running into ye olde Mono 2.0 issue:

http://answers.unity3d.com/questions/858073/bluetooth-support-on-win-and-osx.html

peterfoot, 2017-04-05:
I must admit I don't know how the Mono Sockets stack is implemented but I would have assumed they just call the underlying native winsock functions and therefore that addressfamily would be available on Windows...

Povstalez, 2017-05-14:
I too can't run it in unity3d.
SocketExeption on line `BluetoothClient client = new InTheHand.Net.Sockets.BluetoothClient();`

MeFisto94, 2017-06-06:
I have [investigated](https://stackoverflow.com/a/35038557) this a bit.
There is [this](https://bugzilla.xamarin.com/show_bug.cgi?id=262) Bugreport for Mono which is a Sledgehammer Solution. Even if it would be merged, you cannot use it in Unity since it's running on Mono 2.X and I don't think it's easy to re-compile Mono and especially use it with Unity, especially not on other platforms. If you did that, you still face a lot of issues, which are described by [this](http://blog.mrgibbs.io/pebblesharp-bluez4/), which is linked at the stackoverflow answer.

Now for #6 which means support for Mono **on Linux**, there is the possiblity of using Bluez5 with the DBus which seems to be default on Debian wheezy, so it is a priority to support it anyway.
See [here](http://blog.mrgibbs.io/get-on-d-bus/) for the DBus or [brookpatten/Mono.BlueZ](https://github.com/brookpatten/Mono.BlueZ) for more information.

For Windows, it would be possible to either use PInvoke to have raw access to the Sockets (`ws2_32.dll`) or a more convenient option, considering that the MS Stack only Supports RFCOMM anyway (At least not L2CAP), is just Opening the Virtual COM Port provided by the Driver.

mpb84, 2018-03-08:
Has any progress been made with this? I'm still hoping this can work.

peterfoot, 2018-03-13:
I've been looking into the P/Invoke approach to support Unity but haven't had much time to work on it recently. Given the changes in Windows 10 for UWP apps and a desire to support more platforms where Sockets don't form the basis of Bluetooth comms it's likely to require breaking changes to the API

fashrista, 2018-03-20:
Hey, I am also looking for some bluetooth lib to work with Unity. I'd wery much try to test this project but for the love of god cannot find a built InTheHand.dll file. Tryed to build it myselve but all hell broke loose in visual studio when I tryed to open InTheHand.Net.Personal.sln. I am new to github so is there some way to maybe get allredy built dlls?

peterfoot, 2018-03-20:
Hi @fashrista as mentioned in the thread above the .NET 2.0 version of InTheHand.Net.Personal doesn't work under Unity because Mono sockets throw an exception using the Bluetooth address family. I'm currently looking into a workaround.

peterfoot, 2018-03-22:
An update here on the latest preview version and support for Unity - https://peterfoot.net/2018/03/22/bluetooth-from-unity/

alosaulo, 2018-04-12:
Hi @peterfoot as you mentioned in your blog, you have developed a script for Unity. Would you provide this script as an example to use with the preview package?

cxloyf, 2018-04-20:
hi peterfoot! I also need the script of examlpe  to use with the preview package! Would you provide it ?

peterfoot, 2018-04-23:
It's very minimal but here is an example connecting to a serialport device - https://gist.github.com/peterfoot/ea07a026a6c9d978fd0f27c2431bc745
You can use the DevicePicker to popup a system UI or the DeviceInformation.FindAll to programmatically search and provide your own UI selection

cosmiccricket, 2018-05-22:
Getting a SocketException with error code 0 when I try to use the script posted above in Unity 2017.4.

Miguelotero, 2019-01-12:
Hi @peterfoot . 
Did you finally make any documentation about the preview? Or did you continue developing it?
I am interested in use the library 32feet in Unity.
Thanks in advance.

peterfoot, 2019-01-12:
Still working on it. I'm rolling a number of separate newer spinoff projects into the main codebase to simplify things. The support for Mono (hence Unity) will be there as I've implemented a subset of the Socket APIs directly using the win32 calls. I hope to have a new version available soon and have the GitHub repository re-organised to make it easier to find things.

jafermk, 2019-01-15:
Hi @peterfoot ,
What devices did you use to send the string across, and what platform did you use to read the string. Also, I'm relatively new to Unity and C#, is there a way you could give quick instructions on how I can reproduce your results.
Thank you.

peterfoot, 2020-08-07:
InTheHand.Net.Bluetooth v4.0.8 fully supports Mono (inc Unity)

closed 2020-08-07 by peterfoot

DiemerCohenStuart, 2020-11-06:
Hi @peterfoot, 

I recently downloaded your bluetooth library InTheHand.Net.Bluetooth.4.0.12 (Netstandard 2.0) for Unity in order to list the paired bluetooth devices. I've used the following simple code: 

BluetoothClient client = new BluetoothClient();
BluetoothDeviceInfo device = null;
foreach (var dev in client.PairedDevices)
{
      print(dev.DeviceName);
}

This code seems to work fine in a simple Windows app created with Visual Studio 2019 (.Net Framework 4.7.2), but I can't seem to make it work in the Unity Editor --> client.PairedDevices returns "null". Any suggestions?

We are using Unity version 2019.4.9f1. Target SDK version is 10.0.18362.0. Minimum Platform Version is: 10.0.10240.0.
The final goal is to make it work on a Hololens.

Thank you very much in advance for your support.
Diemer

peterfoot, 2020-11-06:
Hi, not sure how you are extracting the dll from the NuGet package but the .NET Standard dll has no functionality and is just a placeholder. You'll need to use the .NET Framework 4.6.1 version for this to work which should work in recent versions of Unity.

DiemerCohenStuart, 2020-11-06:
Hi, thanks Peter! That worked perfectly. Appreciate the quick reply.