inthehand/32feet #4: No BT actions

cylorijin, 2017-04-30:
Hi!

I have the following setup:

Windows 10 64bit
Windows BT Stack
Visual Studio 2017

I have made the following small test code but it detects 0 devices while Windows itself is detecting some. When i use a BT sniffer i can see there is no BT packets been send from my desktop when i using this code:

`
using System;
using System.Collections.Generic;
using System.ComponentModel;
using System.Data;
using System.Drawing;
using System.Linq;
using System.Text;
using System.Threading.Tasks;
using System.Windows.Forms;

using InTheHand.Net.Bluetooth;
using InTheHand.Net.Sockets;
using InTheHand.Net.Ports;
using InTheHand.Net;
using System.IO;

namespace WindowsFormsApp3 {
    public partial class Form1 : Form {

        public Form1() {
            InitializeComponent();
        }

        private void button1_Click(object sender, EventArgs e) {
            richTextBox1.AppendText("Scanning for devices. Please wait...");
            this.Refresh();
            scan(); 
        }

        private void scan() {
            BluetoothClient client = new BluetoothClient();
            BluetoothDeviceInfo[] devices = client.DiscoverDevices();
            richTextBox1.AppendText(devices.Length.ToString() + " devices found\n");
            BluetoothClient bluetoothClient = new BluetoothClient();
            String deviceAddress;
            String deviceName;
            foreach (BluetoothDeviceInfo device in devices) {
                deviceAddress = device.DeviceAddress.ToString();
                deviceName = device.DeviceName.ToString();
                richTextBox1.AppendText(deviceAddress);
                richTextBox1.AppendText(deviceName);
            }
        }
    }
}
`

peterfoot, 2017-04-30:
A key difference between the 32feet APIs and Windows Settings is that Settings on Windows 10 will display both Low Energy and Classic devices and 32feet only returns Classic devices. Could this be the issue - those devices shown in Windows settings are BTLE devices? If this is not the case can you share a little more detail - what radio does your machine use - built-in or USB dongle, manufacturer etc...

cylorijin, 2017-05-01:
Hi! Many thanks for your prompt reply. You are right. I only have BLE devices. So i cant use the 32feet libary? Is there a similar libary for BLE or is it already build in Windows 10 and VS 2017? I am realy new in this BT thing...

I use a Computer with a USB BT dongle which offers me following devices:
(MS-Stack driver): Generic Bluetooth Radio, MS BT-LE Enumerator, MS BT-Listing
(Manufacturer driver): CSR Bluetooth Radio, 2x Bluetooth SPP-Driver COM Port

This is on my computer for programming and testing. But my goal is it to use my code an a MS tablet.

Thanks in advance
Lars

cylorijin, 2017-05-01:
Ok i have found the Windows Universal App Bluetooth support. :)

peterfoot, 2017-05-01:
Yes you can use the UWP BTLE APIs from a desktop app as well as UWP apps. The 32feet API will work on all desktop Windows versions from Vista onwards but only covers Classic (Rfcomm) connections.

closed 2017-05-01 by peterfoot