inthehand/32feet #7: Is there any way to get the COM port for a ServiceRecord?

Peter-Optiway, 2017-07-26:
Lets say i have the following code:
```C#
var client = new BluetoothClient();
Console.WriteLine("DiscoverDevices");
var devices = client.DiscoverDevices();
foreach(var device in devices)
{
    foreach(var serviceRecord in device.GetServiceRecords(BluetoothService.SerialPort))
    {
        //Get comport for ServiceRecord
    }
}
```

How would i get the Com Port?

peterfoot, 2017-07-26:
Not directly but if you interrogate the registry or use the SetupDI* apis you can determine which COM ports are Bluetooth and which device they map to. There isn't a function in 32feet at the moment to do this.

Peter-Optiway, 2017-07-26:
Thanks for a fast reply.
Where in the registry do i need to look? (and what do i use to match as `ServiceRecord` has no Id?)

peterfoot, 2017-07-26:
Sorry, Registry is for Windows CE only. For desktop Windows you'll need the Setup API. I've put some code into a Gist here:-
https://gist.github.com/peterfoot/b4f61c81023a1e181b9f3940bca344ba

Need to integrate this into a sample app at some time.

Peter-Optiway, 2017-07-26:
That gave me the COM ports im after but i have no idea how to Map each COM port to a `ServiceRecord`.
Im trying to find a comport with a specific name:
[image]
In this case im trying to find the COM port with the name `GNSS Server`, i have been able to find the correct ServiceRecord.

peterfoot, 2017-08-10:
I've updated the gist to add the "RemoteServiceName". This values is stored in the registry and so it's possible to get it based on the device path from those setup APIs. For example on my Zebra printer it returns "Serial Printer". I don't have a device to hand with multiple exposed services but this will give you the string above shown next to the device name e.g. "GNSS Server", "COM1" etc.

Peter-Optiway, 2017-08-11:
That did it, thanks for your help!!!!
One thing if you ever need to use this on your own `RemoteServiceName` contains some trash on the end but the following seems to get rid of it.
`device.RemoteServiceName.TrimEnd((char)0, (char)13, (char)10)`

closed 2017-08-11 by Peter-Optiway

Peter-Optiway, 2017-08-11:
I posted the same question on stackoverflow if you want you can post the same thing there and ill mark it as accepted.
https://stackoverflow.com/questions/45299710/get-bluetooth-comport-name

peterfoot, 2017-08-11:
Thanks, I have posted a response there with a link to the gist