inthehand/32feet #8: Detech Bluetooth LE

etrbet, 2017-08-06:
Hi I develop windows form app with C# and I use 32feet api for bluetooh. I can get Bluetooth devices but I cant get Bluetooth LE devices. Is there any way to get BLE devices on 32feet ?

peterfoot, 2017-08-07:
32feet currently only supports Bluetooth Classic. For Windows 8.1 and upwards you can call the WinRT/UWP API from a desktop app (Windows.Devices.Enumeration etc namespaces)

mahmad123, 2018-10-06:
any plans to add BLE support?

Deluril, 2019-03-19:
G'day. Would there happen to be any update on the introduction of support for Bluetooth LE into the 32feet libraries? I am currently working on a project that is utilising BLE modules and I need to integrate support for connecting to these into a pre-existing Windows Forms app.

fuzioncode, 2019-08-24:
Any updates on this ?  Want to connect to BLE nRF device using .NET Core 2.2 and greater?

peterfoot, 2020-10-01:
InTheHand.BluetoothLE (currently pre-release on NuGet) contains Bluetooth LE functionality. It's still part of 32feet.NET but separate from the Bluetooth Classic API

closed 2020-10-01 by peterfoot