inthehand/32feet #15: Multiple BluetoothService

kirax999, 2017-12-26:
hello, I'm looking for a way to put several Bluetooth Service on the same device example handsFree and phonebook

Do you have an idea ?

peterfoot, 2018-03-13:
You can create multiple listening sockets with different service uuids. However we only support Rfcomm sockets so there isn't the capability built-in to support the audio stream of Handsfree service.

tumelosbk, 2018-10-29:
Humble greetings.

A few years ago I built a c# app that copies the same file to multiple Bluetooth devices simultaneously using multi-threading. I am trying to recreate the same app, but with more features using newer technology(latest 32feet.net version and Visual Studio 2017)

I'm trying to recreate the project from scratch. But I keep getting an exception when I call "ObexWebResponse response = (ObexWebResponse)request.GetResponse();" that Only 1 use of the network/protocol/address can be used at a time.

So should I downgrade the library version? (If that's even possible) Or how can I get passed this issue?
Thanks for the help in advance

closed 2020-02-13 by peterfoot