inthehand/32feet #20: Add Xamarin.Mac via IOBluetooth bindings

peterfoot, 2018-03-18:
Binding project is started but needs review/testing.

lucadesantis, 2018-03-21:
Hello, I'm also trying to write a binding for IOBluetooth in C#. If you want I can help or testing with code, but I not sure we can get results because it is very complex access to system dylib on macOS in C# and the documentation is very poor

peterfoot, 2018-03-21:
I've been able to perform basic operations with the binding I've built so far. The code is checked in here and compiled on NuGet (https://www.nuget.org/packages/InTheHand.IOBluetooth/).

In order to call the API you have to add it and call 
ObjCRuntime.Dlfcn.dlopen("/System.Library/Frameworks/IOBluetooth.framework/IOBluetooth", 0);
I'll need to replace this with a simpler centralised Init method or similar.

So far the basics of the IOBluetoothHostController (local radio device) and IOBluetoothDevice (including PairedDevices property) are working. The next steps are to work on inquiries and rfcomm connections. I've intentionally commented out a lot of the binding code and will add it in as each part is tested and tweaked.

lucadesantis, 2018-03-22:
I used the classic DLLimport always on  "/System.Library/Frameworks/IOBluetooth.framework/IOBluetooth", I worked a on it for short time but I can get the status of Bluetooth (present, on, off).
I am also trying to use the objc_msgSend system call with the selector, but I am finding difficult to understand how format the strings to pass to selector, I'm a beginner in macOS developing

alastor09, 2018-03-29:
I am working on implementing a application to use IOBluetooth.framework and IOBluetoothUI.framework, for selecting a device in xamarin. If you can want can help you. This is the first time i am doing binding so will struggle in start, but surely can learn fast and provide help.

peterfoot, 2018-03-29:
Hi, So far I've generated the binding using ObjectiveSharpie and then began hand editing the resulting binding library to improve some elements like names and using Properties instead of get/set methods. You can see the current code here - https://github.com/inthehand/32feet/tree/master/InTheHand.Devices.Bluetooth/IOBluetooth
This should be a better approach than manually creating P/Invoke definitions for individual functions. I've also started a BluetoothUI library which has a device picker, pairing control and object exchange features. I'll publish a NuGet package for this one too shortly.