inthehand/32feet #26: WidcommDecoratorNetworkStream.cs not found

lopatnov, 2018-04-23:
Error	CS2001	Source file '...\32feet\ITH.Net.Personal.FX4\..\InTheHand.Net.Personal\Net.Bluetooth.Widcomm\WidcommDecoratorNetworkStream.cs' could not be found.

lopatnov, 2018-04-23:
+Error	CS2001	Source file '...\32feet\ITH.Net.Personal.FX4\..\InTheHand.Net.Personal\Net.Bluetooth\BLUETOOTH_DEVICE_INFO.cs' could not be found.

+ Error	CS2001	Source file '...\32feet\ITH.Net.Personal.FX4\..\InTheHand.Net.Personal\Net.Bluetooth\BLUETOOTH_FIND_RADIO_PARAMS.cs' could not be found.

+ Error	CS2001	Source file '...\32feet\ITH.Net.Personal.FX4\..\InTheHand.Net.Personal\Net.Bluetooth\BLUETOOTH_RADIO_INFO.cs' could not be found.

liyulongm, 2018-11-15:
Hi, have you solved you problem?

garykb, 2019-06-25:
I had the same issue.  Got around it by removind WidcommDecoratorNetworkStream.cs from the project.

Added the other 3 files (BLUETOOTH_*) as links from the Net.Bluetooth.Msft folder.

garykb, 2019-07-16:
I'm working on a project that has to run on Windows 7 through Windows 10.  My target is .NET 4.6.1.

I downloaded the repo and added the FX4 build to the InTheHand.Net.Personal solution.  After applying the fixes in my comment above, I was able to build and run the C# packages.  Not much else builds, including the unit test project for the FX4 project.

To get around the build issues, I have a solution file that only builds the FX4 project.

My questions, then are whether anybody else has been able to get a clean build for the FX4 project and can we get the necessary changes incorporated into the root?

Thanks.

peterfoot, 2019-07-23:
There is some work going on in parallel to this to create a new solution which builds in modern versions of VS and removes obsolete code. From Windows 7 the relationship between the Widcomm SDK and Microsoft's own Bluetooth stack and therefore it plugs into the Win32 API so the separate Widcomm dependencies are not required. I'm building the code in a private repository because it involves breaking and moving a lot of the code files but when it is complete enough to be shared I'll move it into this repository alongside the legacy codebase. This work will also make it easier to broaden the platform support. While Win32 is the priority I'm hoping to get Android, Linux, macOS, UWP and others into the mix.

garykb, 2019-07-23:
That sounds good.  I'll hold off doing a pull request for my fixes then.

Thanks.

closed 2019-08-07 by lopatnov