inthehand/32feet #29: PairRequest always return false when I try to pair bluetooth mouse

hchtao, 2018-08-07:
Hi,
   I am using 32feet to discover a bluetooth mouse and try to pair and connect to it, but the PairRequest always return false.  The device class is "miscellaneous". The function HandleRequests is not executed too.
   The following is my code.

EventHandler<BluetoothWin32AuthenticationEventArgs> handler = new EventHandler<BluetoothWin32AuthenticationEventArgs>(HandleRequests);
BluetoothWin32Authentication auth = new BluetoothWin32Authentication(handler);

 protected void PairBluetoothTask(BluetoothDeviceInfo device)
 {
            bool isPaired = BluetoothSecurity.PairRequest(device.DeviceAddress, null); 
            if (isPaired)
            {
                Console.WriteLine(device.DeviceName + " paired!");
            }
            else
            {
                Console.WriteLine(device.DeviceName + " not paired!");
            }
}

private void HandleRequests(object that, BluetoothWin32AuthenticationEventArgs e)
{
            e.Confirm = true;
}

peterfoot, 2018-08-07:
What OS version are you running? Are you sure that an empty pin is the expected value?

hchtao, 2018-08-07:
I am using Win10, I think it is because my Bluetooth mouse is BLE. May I know whether 32feet support BLE now?

peterfoot, 2018-11-15:
At the moment this api doesn’t support pairing with BTLE devices. I’m currently working on integrating this.

closed 2018-11-15 by peterfoot