inthehand/32feet #31: Unity Android/Windows

w00dst0ck8786, 2018-08-23:
I am right now trying to create an application that is basically a Unity Android remote talking to a Unity PC application via Bluetooth.

It is a first sketch as we actually only need a few buttons on the Android side to trigger certain actions on the PC side.

At first I built all of this successfully with a WiFi connection and then tried to convert it to a Bluetooth connection.
I found your library and integrated it and am stuck now, because of the socket problem you describe on your blog. I was so glad you already addressed that problem and I downloaded the new libs.

Now I am a bit at a loss because the library and its classes seem to be a lot different.
I originally created my applications with a BluetoothListener and a BluetoothClient. Both seem not to be present anymore in the library.

Maybe you have a small sample project you could provide where one can see how it works now? I was so happy when I saw you addressed the Unity issue and I think I am so close with my app. 

I would be really thankful for some help.

wywarren, 2018-09-01:
Could you explain how you got the libraries into Unity in the first place? I tried using nuget on my Unity project sln and got a 

`The project does not support adding package references through the add package command.`

error message.

Opening the solutions included in this git on Visual Studio 2017 also yields a bunch of errors on as it tries to update the solution to the latest version.

w00dst0ck8786, 2018-09-04:
You can import DLLs in Unity, no problem. I downloaded the package and put the DLLs directly into the Unity project, into the Assets in Unity.
https://docs.unity3d.com/Manual/UsingDLL.html

I have worked with libraries in the past, also with Windows and Android libraries in one project. Unity is pretty well setup and it works. 

But I got the socket error and with the new version...I have no idea ow to use it.

peterfoot, 2018-09-04:
The dll in the net35 folder has the workaround for Sockets. I'm looking at integrating two APIs so there is both the old Sockets API and the newer API which supports BT Classic and BTLE in the same dll.

w00dst0ck8786, 2018-09-04:
@peterfoot  
Yes I used that dll, after I read about the problem on your blog. I just wrote my code like the samples suggested and this doesn't work anymore. So thank you for looking into that. I am really curious if the Unity stuff will work

wywarren, 2018-09-05:
I ended up putting the dll from the net35 folder into Unity so I'm able to use InTheHand.Devices.Bluetooth now. However, some of the examples I found using feet32 require InTheHand.Net.Sockets which I think you described in your blog about it not currently working inside unity. Where can I find sample code for Unity that only uses InTheHand.Devices.Bluetooth? I don't seem to see anything resembling that in the samples folder.

wywarren, 2018-09-05:
Using the dll from the net35 folder, I fail to get devices using:
`var devices = DeviceInformation.FindAll(RfcommDeviceService.GetDeviceSelector(RfcommServiceId.SerialPort));`

The `RfcommDeviceService.GetDeviceSelector(RfcommServiceId.SerialPort)` portion sees my bluetooth device but the FindAll returns an empty list. Also FindAllAsync is not wrapped into this DLL as well so I haven't tried that one either.

izukingz, 2019-01-20:
Hello Peter, are you still working on the new API for unity. I'm also looking for an example on how to use InTheHand.Devices.Bluetooth.

SteenPetersen, 2019-10-26:
I would greatly appreciate some more information on how to use InTheHand.Devies.Bluetooth as well. I am trying to simply discover bluetooth devices as a way "hello world".

izukingz, 2019-10-26:
@SteenPetersen, what I eventually did was setup my arduino device to use serial communication. Then on PC I used this serial comm asset ardity https://assetstore.unity.com/packages/tools/integration/ardity-arduino-unity-communication-made-easy-123819. You have to first pair the BT device using Windows add BT device. After that go to device manager to get the COM port. This is what you will use in ardity to connect to and communicate. You can also get all com ports through code and use a strategy to discover which one is yours.

izukingz, 2019-10-26:
For some reason I thought this was for unity, anyway I believe ardity should still work outside unity. It's a c# project. You just need to figure out how to use it without the ui part.

SteenPetersen, 2019-10-26:
Thanks, I’ll look into it. Getting some [sensors](https://www.amazon.com/Accelerometer-Triple-Axis-Acceleration-Transducer-BWT61/dp/B018NL1R0Y) soon that I need to make communicate with unity android builds. Trying to figure out how to go about it.

izukingz, 2019-10-27:
@SteenPetersen , Note that Ardity is for PC serial communictaion using unity. For Android/iOS asset e.g https://assetstore.unity.com/packages/tools/input-management/android-microcontrollers-bluetooth-16467

Also note that your sensor has a tutorial that you can download from here: https://drive.google.com/file/d/1Y3Hs8RiAHhglAU0FOO9UwfwT7eForKgA/view
They even have code samples too though not from unity.

closed 2020-02-13 by peterfoot