inthehand/32feet #35: Bonded devices are not retained on Windows 10

eamreyes, 2018-09-20:
I'm using classic Bluetooth to communicate between an Android app and a WPF application. The initial pairing process has been a major pain point when onboarding new users. My plan to simplify the pairing process is to provide custom dialogs inside the app (via InTheHand.Net.Bluetooth.BluetoothWin32Authentication). However, devices paired in this way are not retained by the Windows 10 system. Every connection leads to a pairing request.

1. Connect from Android app directly to the desktop using the Bluetooth device address
2. If the devices haven't been paired, Android initiates a pairing request.
3. The desktop uses the  InTheHand.Net.Bluetooth.BluetoothWin32Authentication and a custom dialog to present the pairing passcode and allow the user to confirm.
4. I complete the pairing process on both devices, then the connection is establish and the applications communicate properly.
5. Close the connection
6. Try to re-establish the connection from the Android app.

- From the Android OS perspective the devices are paired/bonded. But the pairing dialog still pops up.
- In Windows 10 the device does not appear in the list of known Bluetooth devices.
- If I ignore my custom dialog, and use the Windows 10 pairing dialog that pops up, the pins do not match but the devices pair as expected. A pairing/bonding using this method does not require to be paired after the first pairing.

Is this expected behavior for the library? Thoughts, advice?

PritamUpadhyay, 2019-05-17:
Hi @eamreyes , 
Have you been able to establish the communication between your Android app and WPF application using Classic Bluetooth connection?

peterfoot, 2019-05-17:
Which version of Windows 10 are you running on? And do you know which pairing method is being used (in the BluetoothWin32Authentication callback)?

eamreyes, 2019-05-21:
> Hi @eamreyes ,
> Have you been able to establish the communication between your Android app and WPF application using Classic Bluetooth connection?

Yes. Using a manual pairing process on W7, W8, W10, allowed establishing communication between Android app and WPF application.

> Which version of Windows 10 are you running on? And do you know which pairing method is being used (in the BluetoothWin32Authentication callback)?

I do not recall. I'm no longer working on the project. If I could do it all over, I would probably use BLE if the APIs were available.

closed 2020-02-13 by peterfoot

PritamUpadhyay, 2020-03-09:
@eamreyes , thanks.