inthehand/32feet #37: Support BlueZ 5

peterfoot, 2018-10-25:
BlueZ 5 introduces changes to the API so will require code changes:-
http://www.bluez.org/bluez-5-api-introduction-and-porting-guide/

Opened as separate issue from #6

werddomain, 2020-04-28:
is there any update about this?
Like .net core on linux?

peterfoot, 2020-04-29:
Ideally I need some help adding new platforms. Please let me know if you'd like to contribute a Linux provider.

werddomain, 2020-04-29:
When i find a way to use bluetooth on raspberry pi in c#, i will be able to help this project ...

asheesh1996, 2021-12-13:
Are there any updates on this issue.

DamianSuess, 2023-06-23:
@peterfoot thank you for supporting the [Linux.Bluetooth](https://github.com/SuessLabs/Linux.Bluetooth) project as part of 32feet!  It's truly an honor, as I started using your library 10+ years ago, and it inspired me to build this.

Please let me know if there are any API-level changes that would make support for your project easier. I'd be happy to work with you. 👍

peterfoot, 2023-06-29:
Thanks, it provides a great API over the DBus workings. I didn't want to reinvent the wheel so am trying to incorporate it so it's another platform supported with the same cross-platform API (which itself is based on WebBluetooth). I've not had a lot of time to work on it so it's still a work in progress.

peterfoot, 2023-06-29:
@DamianSuess if I recall, Linux.Bluetooth doesn't currently support Descriptors? At some point it would be useful to add those if they are not currently in progress...

DamianSuess, 2023-06-29:
@peterfoot, I'll get it on the backlog and let's make it happen

whSwitching, 2023-08-16:
#149  take me here,  any progress on this?

closed 2024-03-09 by peterfoot