inthehand/32feet #39: newb: SocketException An invalid argument was supplied

yisman, 2018-11-14:
Hi
I'm totally new to bluetooth and 32feet
I have a few devices that windows 7 recognizes and pairs correctly (though i havent progressed further then the pairing)
but in 32feet, all devices cause the following error

System.Net.Sockets.SocketException: An invalid argument was supplied 103047170F76:xxxxxxxx....

the error code is 10022

and here is the stacktrace

   ```
at System.Net.Sockets.Socket.DoConnect(EndPoint endPointSnapshot, SocketAddress socketAddress)
   at System.Net.Sockets.Socket.Connect(EndPoint remoteEP)
   at InTheHand.Net.Bluetooth.Msft.SocketBluetoothClient.Connect(BluetoothEndPoint remoteEP) in C:\Users\CEO\Downloads\32feet-master\32feet-master\InTheHand.Net.Personal\Net.Bluetooth.Msft\SocketBluetoothClient.cs:line 698
   at InTheHand.Net.Sockets.BluetoothClient.Connect(BluetoothEndPoint remoteEP) in C:\Users\CEO\Downloads\32feet-master\32feet-master\InTheHand.Net.Personal\Net.Sockets\BluetoothClient.cs:line 414
   at InTheHand.Net.Sockets.BluetoothClient.Connect(BluetoothAddress address, Guid service) in C:\Users\CEO\Downloads\32feet-master\32feet-master\InTheHand.Net.Personal\Net.Sockets\BluetoothClient.cs:line 447
   at Chat2DesktopVb.Form1.BluetoothConnect(BluetoothAddress addr) in C:\Users\CEO\Downloads\32feet-master\32feet-master\Samples\BluetoothChat2\Chat2DesktopVb\Form1Core.vb:line 73
   at Chat2DesktopVb.Form1.menuItemConnectBySelect_Click(Object sender, EventArgs e) in C:\Users\CEO\Downloads\32feet-master\32feet-master\Samples\BluetoothChat2\Chat2DesktopVb\Form1Core.vb:line 290
   at System.Windows.Forms.ToolStripItem.RaiseEvent(Object key, EventArgs e)
   at System.Windows.Forms.ToolStripMenuItem.OnClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleClick(EventArgs e)
   at System.Windows.Forms.ToolStripItem.HandleMouseUp(MouseEventArgs e)
   at System.Windows.Forms.ToolStrip.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.ToolStripDropDown.OnMouseUp(MouseEventArgs mea)
   at System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ToolStrip.WndProc(Message& m)
   at System.Windows.Forms.ToolStripDropDown.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.DebuggableCallback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
   at System.Windows.Forms.UnsafeNativeMethods.DispatchMessageW(MSG& msg)
   at System.Windows.Forms.Application.ComponentManager.System.Windows.Forms.UnsafeNativeMethods.IMsoComponentManager.FPushMessageLoop(Int32 dwComponentID, Int32 reason, Int32 pvLoopData)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoopInner(Int32 reason, ApplicationContext context)
   at System.Windows.Forms.Application.ThreadContext.RunMessageLoop(Int32 reason, ApplicationContext context)
   at Microsoft.VisualBasic.ApplicationServices.WindowsFormsApplicationBase.OnRun()
   at Microsoft.VisualBasic.ApplicationServices.WindowsFormsApplicationBase.DoApplicationModel()
   at Microsoft.VisualBasic.ApplicationServices.WindowsFormsApplicationBase.Run(String[] commandLine)
   at Chat2DesktopVb.My.MyApplication.Main(String[] Args) in :line 81

```

I tried stopping and starting the bluetooth service in windows
also device manager shows that the drivers for both bluetooth devices (genereic bluetooth radio and microsoft bluetooth enumerator)  are up to date

what may thge issue be?

thanks

peterfoot, 2018-11-14:
Hi, that's not a common error message and I can't see what exactly is happening. Just prior to this you selected Connect by select and then selected a device which is running the same Chat2Desktop app?

yisman, 2018-11-14:
hi
thanks for answering.
and sorry, maybe thats my mistake
i just selected an arbitrary phone device
if this sample is not for this, can you point me to a .net sample which
grabs caller id from an incoming call on a paired bluetooth device?
thanks a lot

-- 

*Sincerely,*

*Yisroel M. Olewski          *

*CEO*

Machshevet.com

+972 (52) 764-9971

*Websites **·** CRM **·** ERP **·** Telephony **·** Inventory **·** Content
Management*

התכתבות עם כתובת מייל זו כשולח או נמען/מצורף מהווה אישור לקבלת דואר פרסומי
במייל. ניתן להפסיק את הדיוור בלחיצת על unsubscribe בגוף המייל הפרסומי

peterfoot, 2018-11-14:
There aren't any samples for the phone profiles currently. You'd have to implement based on the Bluetooth spec for the profile.

The Bluetooth chat one will only work if you load the app onto another device and connect to it, the picker doesn't filter out devices which are not advertising the chat service.

yisman, 2018-11-15:
Hi
I really appreciate your input
Is there a .net sample which connect to any arbitrary device?
I guess once im connected i can somehow read the incoming data and parse it?
I hope i make sense since im totaly new to bluetooth
Thanks again

Yisroel M. Olewski
Machshevet.com
Tel. +972 527649971

yisman, 2018-11-27:
hi
if possible i would appreciate a sample which can connect to any mobile
device and read the caller id info when it rings
i could not seem to find any such sample in the samples folder
google searches didnt bring anything useful, either
i'd be realy grateful if you can help out
all the best

-- 

*Sincerely,*

*Yisroel M. Olewski          *

*CEO*

Machshevet.com

+972 (52) 764-9971

*Websites **·** CRM **·** ERP **·** Telephony **·** Inventory **·** Content
Management*

התכתבות עם כתובת מייל זו כשולח או נמען/מצורף מהווה אישור לקבלת דואר פרסומי
במייל. ניתן להפסיק את הדיוור בלחיצת על unsubscribe בגוף המייל הפרסומי

peterfoot, 2018-11-27:
At the moment that is out of scope as it goes beyond the 32feet functionality and requires implementation of the Headset or Handsfree profiles over a connection. If you require this for a commercial project we can quote for the development.

yisman, 2018-11-27:
Ah
So it cannot be done at all currently?
Yes, its commercial. What would you charge?

Yisroel M. Olewski
Machshevet.com
Tel. +972 527649971

peterfoot, 2018-11-27:
It can be done. The Headset profile uses an RfComm channel for command and control and a separate SCO channel for audio. 32feet doesn't provide support for the audio portion. In order to send and receive commands to a Headset device you need to following the relevant Bluetooth spec to build and parse the data packets used. The same is true of many RfComm based profiles. Some of the commands are borrowed from Modem specs so AT RING represents an incoming call. This should be accompanied by display number depending on the capabilities of the connected device.

yisman, 2018-11-27:
im afraid its a bit over my head
if you can do the implementation, i will fowrard the price to my client for
approval
thanks

-- 

*Sincerely,*

*Yisroel M. Olewski          *

*CEO*

Machshevet.com

+972 (52) 764-9971

*Websites **·** CRM **·** ERP **·** Telephony **·** Inventory **·** Content
Management*

התכתבות עם כתובת מייל זו כשולח או נמען/מצורף מהווה אישור לקבלת דואר פרסומי
במייל. ניתן להפסיק את הדיוור בלחיצת על unsubscribe בגוף המייל הפרסומי

closed 2020-02-13 by peterfoot

yisman, 2021-09-02:
Make your new year an innovative one!

View this email in your browser (https://mailchi.mp/machshevet/drfklqb6ix-4666046?e=41bbaa989c)
https://machshevet.us3.list-manage.com/track/click?u=752c5e4f79a9fa7c8f2e896ea&id=e1f771c482&e=41bbaa989c
בס"ד

לכל לקוחותיי וידידיי היקרים

[ אין לכם זמן לקרוא את ההודעה?
פנו אלינו לקבל מערכת חדשה לניהול נתונים, שתחסוך לכם שעות כל יום!! ]

בפרוס השנה החדשה הבאה עלינו לטובה,
נאחל לכם ולמשפחתכם
ברכת שנה טובה ומתוקה,
שנת בריאות, שמחה ופרנסה טובה,
שפע ברכה והצלחה.
Wishing you all a happy and healthy New Year

תוסף חדש מבית 'מחשבת'
מהיום תראה את כל פרטי אנשי הקשר בחלון ה-Gmail שלך.
https://machshevet.us3.list-manage.com/track/click?u=752c5e4f79a9fa7c8f2e896ea&id=bd2ab5d16f&e=41bbaa989c
Turbo Paced Custom Software Development
Tel: +972 (52) 764-9971 (tel:972527649971)

============================================================
** (https://machshevet.us3.list-manage.com/track/click?u=752c5e4f79a9fa7c8f2e896ea&id=399eeb5ecc&e=41bbaa989c)
** ***@***.***)
Copyright © 2021 machshevet, All rights reserved.
 contact

Our mailing address is:
machshevet
88 Agripas Street
jerusalem 9453331
Israel
Want to change how you receive these emails?
You can ** update your preferences (https://machshevet.us3.list-manage.com/profile?u=752c5e4f79a9fa7c8f2e896ea&id=37a3cbc8b7&e=41bbaa989c&c=a6138bfd9a)
or ** unsubscribe from this list (https://machshevet.us3.list-manage.com/unsubscribe?u=752c5e4f79a9fa7c8f2e896ea&id=37a3cbc8b7&e=41bbaa989c&c=a6138bfd9a)
.
 Email Marketing Powered by Mailchimp
http://www.mailchimp.com/email-referral/?utm_source=freemium_newsletter&utm_medium=email&utm_campaign=referral_marketing&aid=752c5e4f79a9fa7c8f2e896ea&afl=1

yisman, 2022-09-20:
Make your new year an innovative one!

View this email in your browser (https://mailchi.mp/machshevet/drfklqb6ix-5231876?e=41bbaa989c)
https://machshevet.us3.list-manage.com/track/click?u=752c5e4f79a9fa7c8f2e896ea&id=95f7cedbf2&e=41bbaa989c
בס"ד

לכל לקוחותיי וידידיי היקרים

[ אין לכם זמן לקרוא את ההודעה?
פנו אלינו לקבל מערכת חדשה לניהול נתונים, שתחסוך לכם שעות כל יום!! ]

בפרוס השנה החדשה הבאה עלינו לטובה,
נאחל לכם ולמשפחתכם
ברכת שנה טובה ומתוקה,
שנת בריאות, שמחה ופרנסה טובה,
שפע ברכה והצלחה.
Wishing you all a happy and healthy New Year

מיוחד מבית "מחשבת"
אזור איש ללקוחות שלכם לראות סטטוס הזמנות, חיובים ועוד
https://machshevet.us3.list-manage.com/track/click?u=752c5e4f79a9fa7c8f2e896ea&id=239a4155f1&e=41bbaa989c
Turbo Paced Custom Software Development
Tel: +972 (52) 764-9971 (tel:972527649971)

============================================================
** (https://machshevet.us3.list-manage.com/track/click?u=752c5e4f79a9fa7c8f2e896ea&id=21034de83b&e=41bbaa989c)
** ***@***.***)
Copyright © 2022 machshevet, All rights reserved.
 contact

Our mailing address is:
machshevet
88 Agripas Street
jerusalem 9453331
Israel
Want to change how you receive these emails?
You can ** update your preferences (https://machshevet.us3.list-manage.com/profile?u=752c5e4f79a9fa7c8f2e896ea&id=37a3cbc8b7&e=41bbaa989c&c=1f3e2f8a3b)
or ** unsubscribe from this list (https://machshevet.us3.list-manage.com/unsubscribe?u=752c5e4f79a9fa7c8f2e896ea&id=37a3cbc8b7&e=41bbaa989c&c=1f3e2f8a3b)
.
 Email Marketing Powered by Mailchimp
http://www.mailchimp.com/email-referral/?utm_source=freemium_newsletter&utm_medium=email&utm_campaign=referral_marketing&aid=752c5e4f79a9fa7c8f2e896ea&afl=1