inthehand/32feet #40: Platform not supported exception in Unity : 32feet.NET not supporting bluetooth stack

rabhattaraiL, 2018-11-29:
I have been trying to connect to BioRadio device through bluetooth. My code works on visual studio and collects the required data after bluetooth connection. When I try to run it from Unity, it does not work and gives the exception error of 32feet.NET.  I did add the nuget package from the references tab in visual studio but it still shows the same error. I am new to Unity and using bluetooth through visual studio. 
Would anyone help to understand step by step how to resolve it ?

PocketDoor, 2019-01-14:
I am also having issues with "PlatfornNotSupportedException" in a similar situation.

jcs090218, 2020-07-16:
Any progress or solution on this?

peterfoot, 2020-07-16:
Are you using the current InTheHand.Net.Bluetooth 4.0.6 release?

jcs090218, 2020-07-17:
@peterfoot Who were you referring to? :)

I have not try this package yet, and I am just looking for possible development obstacle. I wonder if this problem are resolved because I am about to use this package with Unity and build to other platform! 😖 Thanks! 😄

peterfoot, 2020-08-07:
There was actually a bug that was resolved in 4.0.8 to fix connection issues on Mono (inc Unity). All functionality should now be enabled inside Unity

closed 2020-09-26 by peterfoot