inthehand/32feet #41: Receiving data from SPP profile fails under Windows 10 Version 1809

HMprimes, 2018-12-03:
We are using the 32feet library to connect to a Bluetooth Classic module (Bluegiga/Silabs WT11) via a virtual COM port connection (SPP). 
Below is an excerpt of the C# code we use.

## Code

```
...
// connect
InTheHand.Net.Sockets.BluetoothClient m_localClient;
m_localClient = new InTheHand.Net.Sockets.BluetoothClient();
m_localClient.Connect(mapping.Address, BluetoothService.SerialPort);
...

// receive data
if (m_localClient != null && m_localClient.Connected){
   Stream inStream = null;
   inStream = m_localClient.GetStream();
   int numberOfBytes = inStream.Read(m_receivedData, 0, m_receivedData.Length);
   ...
}   
...
// send data
...
if (m_localClient != null && m_localClient.Connected){
   Stream outStream = m_localClient.GetStream();
   if (outStream != null){
      outStream .WriteTimeout = 1000;
      outStream .Write(data, 0, data.Length);
   }
}
...
// disconnect
if (m_localClient != null){
    m_localClient.Close();
    m_localClient.Dispose();
    m_localClient = null;
}
```
## Problem 

Under Windows 7 this code works.
Under Windows 10 (1803 Build 17134) the code throws an exception at the receive data part.
We are still running tests with earlier versions of Windows 10 to see when the problem was introduced because there are indications that with earlier versions of Windows 10 (1507 or higher) the code worked. If we find out which versions work and which not, I will update this issue accordingly.

Connecting the virtual COM ports created by the Windows Bluetooth stack via any terminal software works under Windows 10 as well as under Windows 7, so there is no general problem with using the SPP profile.

Interestingly on the Bluetooth module side we receive the message "SYNTAX ERROR" on the serial output when trying to establish the connection using the 32feet library code above. This error does again not 

## Error

The error we get ist mostly this one:
System.IO.IOException: Unable to read data from the transport connection: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond. ---> System.Net.Sockets.SocketException: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond
   at System.Net.Sockets.NetworkStream.Read(Byte[] buffer, Int32 offset, Int32 size)

Less often this one 
System.Net.Sockets.SocketException (0x80004005): An invalid argument was supplied 00078025BEDA:0000110100001000800000805f9b34fb
   at System.Net.Sockets.Socket.DoConnect(EndPoint endPointSnapshot, SocketAddress socketAddress)
   at System.Net.Sockets.Socket.Connect(EndPoint remoteEP)
   at InTheHand.Net.Bluetooth.Msft.SocketBluetoothClient.Connect(BluetoothEndPoint remoteEP)
   at InTheHand.Net.Sockets.BluetoothClient.Connect(BluetoothAddress address, Guid service)

Or that one:
System.Net.Sockets.SocketException (0x80004005): An operation was attempted on something that is not a socket 00078025BEDA:0000110100001000800000805f9b34fb
   at System.Net.Sockets.Socket.DoConnect(EndPoint endPointSnapshot, SocketAddress socketAddress)
   at System.Net.Sockets.Socket.Connect(EndPoint remoteEP)
   at InTheHand.Net.Bluetooth.Msft.SocketBluetoothClient.Connect(BluetoothEndPoint remoteEP)
   at InTheHand.Net.Sockets.BluetoothClient.Connect(BluetoothAddress address, Guid service)

peterfoot, 2018-12-03:
Is the device successfully paired on your Windows 10 machine? What Bluetooth radio does your PC have? You can view Device Manager and look under the Bluetooth node. You should see a Bluetooth Device (the radio) and Microsoft Bluetooth Enumerator and Microsoft Bluetooth LE Enumerator along with paired devices.
You should also see the device in Settings >Devices > Bluetooth & other devices and it's state should change from Paired to Connected when your app established the connection.

HMprimes, 2018-12-03:
I intended to add more info to the initial issue before submitting but accidentally found out there is a keyboard shortcut for submitting, so the issue got submitted before I finished writing.

Yes, the device is successfully paired with Windows.
I have Microsoft Bluetooth Enumerator and Microsoft Bluetooth LE Enumerator as well as the paired device showing up. Im not 100% sure about the change from "Paired" to "Connected" in "Settings >Devices > Bluetooth & other devices". There seems to be a connection for a short time but then we are back to paired. When I open the COM port (COM11 out of COM3 and COM 11) in this case port via Putty we definitely geht a connected in "Settings and Devices".

We tested on multiple machines with multiple devices and dongles. Currently I am using an CSR8510 A10 dongle (VID_0A12&PID_00019, Microsoft Driver Version 10.0.17134.376) but the behaviour is the same with two other dongles we tested.

Im currently speculating that there might have been a change in some kind of authentication with Windows 10 Version 1083 which our modules fail to respond to. Also the modules specify SPP 1.1 compatibility while Windows 10 now uses SPP 1.2, but to my knowledge this should not be an issue.

peterfoot, 2018-12-03:
Does the device require Authentication/Encryption and do you set the relevant Authenticate/Encrypt properties on the BluetoothClient before the Connect() call. It is possible that Windows 10 has somehow changed the defaults for these. Perhaps this explains the syntax_error on the remote device?

HMprimes, 2018-12-03:
We don't require authentication on the device side. However, I indeed think that from a certain Windows 10 version something authentication related has changed, which is somehow causing the problem. Unfortunately Microsoft is not exactly known detailed changelogs or at least pretty good in hiding them from me.

After some more looking into, it seems that the 32feet library in Windows 10 establishes the connection and it is immediately closed again. If this happens by the  32feet library, by Windows or by the module is not clear yet.

Windows 10 1083 is also affected. We are still testing older versions.

The closing seems to happen almost always in between the test for  "m_localClient.Connected == true" and finishing the actual stream reading "int numberOfBytes = inStream.Read(m_receivedData, 0, m_receivedData.Length);". The errors we get are thus because the stream reader looses its connection in the middle of reading data. Sometimes a the connection is lost a bit sooner, sometimes later, because we get different amounts of data out. The first transfers involved are one packet of 515 bytes and one packet of 512 bytes. Mostly the first transfer already fails, but sometimes we get into the second one, but latest then the connection is gone.

The "SYNTAX ERROR" I mentioned above as given out by the Bluetooth module seems to be related to the fact that our device code, depending on the timing" has not yet realized, that the connection is lost and tries to send data to a link ID that is no longer connected.
This behaviour is described in the UG218 (https://www.silabs.com/documents/login/user-guides/UG218.pdf) as a known issue (IWRAP821):  “Ignored commands called just after closing Bluetooth connection. iWRAP commands may be ignored or may cause SYNTAX ERROR if called just after closing connection. No problems were found if commands are called 1 second or later after connection close.”
So this error is actually a result of the disconnect and not the cause.

HMprimes, 2018-12-06:
The problem is "fixed".

I am documenting the "fix" here, in case someone else experiences the same problem.

The Exceptions were caused by setting a timeout on the instream, which for some unknown reason I did not include in the read code, but only for the write code above.

## Solution 
Removing the timeout from the reading stream, resulting in a default value of "infinite" solves the issue.

## Cause
Our code sets a default timeout of 100 ms. 

`
if (m_localClient != null && m_localClient.Connected){
 …
 inStream = m_localClient.GetStream();
 inStream.ReadTimeout = 100;
 …
 inStream.Read(m_receivedData, 0, m_receivedData.Length);
 …
}
`

The above code works with Windows 7, with our default 100 ms timeout.

The above code does not work with Windows 10. Even if the timeout is increased to 1000 ms it fails, at which point we stopped looking into the timeout as the source of the problem, but rather focussed on the fact that the Bluetooth connection gets closed.

Seemingly Windows 10 takes a lot longer than Windows 7 to establish the Bluetooth stream to a connected , after the 32feet library states it is connected. A timeout of 10000 ms or no timeout at all on the stream is working.

## "Affected" Windows 10 versions
Below also the results for the different Windows 10 versions. All the versions we tested were showing the same problem.

1507 Build 10240 affected
1511 Build 10586 not tested
1607 Build 14393 not tested
1607 Build 14393 affected
1703 Build 15063 affected
1709 Build 16299 affected
1803 Build 17134 affected
1809 Build 17763 affected

@peterfoot: Thanks taking the time to look into this.

closed 2018-12-06 by HMprimes