inthehand/32feet #51: InTheHand.Devices.Bluetooth any complete examples ?

pirhano, 2019-06-26:
I could find any examples related to the new API

Shamalmanoj, 2019-08-08:
https://archive.codeplex.com/?p=32feet

matthewjones555, 2019-08-15:
@pirhano, I'm also looking at trying to get a working sample up and running, and what I've found in the documentation does not match up with what I'm seeing in the nuget package.  It's very confusing.

I'm trying to follow [this example](https://github.com/inthehand/32feet/wiki/Discovery) on device discovery:

    var cli = new BluetoothClient();
    BluetoothDeviceInfo[]() peers = cli.DiscoverDevices();
    BluetoothDeviceInfo device = null;//= ... select one of peer()...
    BluetoothAddress addr = device.DeviceAddress;

But I can't find any of the classes in the nuget package.  I can find them in the repository, but it seems to be a completely different API than the one exposed on nuget.

The code looks like exactly what I'm looking for, but I've no idea where the classes are located.

😕

johnson512n28, 2019-11-13:
To @matthewjones555's comment, the documentation available all seems to be for the old API.. Both the BluetoothListener and BluetoothClient classes have been replaced. The v4 package is still in preview, but but it would be nice to have an article/blog post about what exactly is changing, and how to update from using these two classes.

peterfoot, 2019-11-14:
Sorry for the delay on this. I've stopped work on the "new" InTheHand.Devices.Bluetooth API and instead have a new version modelled on the original API but adding .NET Standard support and better support for NuGet, Unity and eventually other platforms. I hope to have the new code merged in soon along with a new v4.0 NuGet package.

matthewjones555, 2019-11-15:
Thanks for the update @peterfoot.  I'll keep an eye out for any updates.

tomitrescak, 2020-08-03:
Hello, thanks for the package. Is there any update on Unity support? Thanks!

peterfoot, 2020-08-03:
InTheHand.Net.Bluetooth is out on NuGet. It uses Win32 Socket APIs to get around the issue where Mono Sockets don't allow the Bluetooth namespace/address family. The API is closely modelled on the original 32feet.NET v3.x API with some simplifications.

tomitrescak, 2020-08-03:
@peterfoot thanks for your answer! I used the library you recommended, but it does not process LE devices. When I tried to use the LE library in my project it would not compile as it was missing "Windows.Foundation.UniversalApiContract". I really like the BluetoothPicker approach, but unfortunately it only discovers standard BT devices.

This is my code:

```csharp
var pick = new BluetoothDevicePicker();
pick.PickSingleDeviceAsync().ContinueWith(task =>
{
            var dev = task.Result;
            Debug.Log(dev.DeviceName);
});
```

peterfoot, 2020-08-07:
For Bluetooth LE there is a separate library - InTheHand.BluetoothLE. I need to do a new release and update documentation/samples as it's a completely different API

tomitrescak, 2020-08-12:
Awesome! Can’t wait to try it in unity! Will make our life so much easier ;)

AsfandNadeem, 2021-08-30:
When i try to connect using IntheHand.net.bluetooth Bluetoothclient. I get an exception.
`client.BeginConnect(endPoint, new AsyncCallback(ConnectCB), client)
`
I get null reference exception.

What might be causing the exception?