inthehand/32feet #53: Connecting only one device without searcing

ifurkanvural, 2019-07-24:
I need a support about connecting Bluetooth device. 

My purpose is to communicate only one Bluetooth device. I don’t wanna to make search again and again. 

  private void send_file ()
        {
            InTheHand.Net.Sockets.BluetoothClient bc = new InTheHand.Net.Sockets.BluetoothClient();
            InTheHand.Net.Sockets.BluetoothDeviceInfo[] dizi = bc.DiscoverDevices();

                for (int i = 0; i < dizi.Length; i++)
                {
                    if (dizi[i].DeviceName == "DESKTOP-OR4D5FK")
                    {
                    
                    System.Uri uri = new Uri("obex://" + dizi[i].DeviceAddress + "/" + "0.jpg");
                        // System.Uri uri = new Uri("obex://" + "DESKTOP-OR4D5FK" + " / " + "deneme.txt");
                        ObexWebRequest request = new ObexWebRequest(uri);
                        request.ReadFile("0.jpg");/////
                        ObexWebResponse response = (ObexWebResponse)request.GetResponse();
                        response.Close();
                    break;
                    }
                }
        }

Here is my example code. 

I just only want to sent a file to specific device. 

Kindly waiting your help

peterfoot, 2019-07-24:
Okay, that basically looks right - you can store a device address and reuse it without discovering every time. What is not working?

Peter

ifurkanvural, 2019-07-24:
Dear Peterfoot,

Actually, it is working as you said. But I am always waiting discoverdevices() method. So it is taking time to find all devices. I don't wanna discover again. I just want to send a file to specific device.

Thank you.

peterfoot, 2019-07-24:
You can store the addresses received from a discovery and then use them again - just make sure you handle the scenario where the device is no longer available and the connection will fail. BluetoothAddress can be persisted as a long/ulong or string in memory or in a file or database and then used later.

ifurkanvural, 2019-07-25:
Dear Peter,
If you have a little time, could you please edit my code ?
https://drive.google.com/open?id=17DoOv00oMkFiNtOkDhpXHlR_3qsHYFzX

ifurkanvural, 2019-08-25:
Hey Peter,

I am very in bad situation. Please help me about below topic. 

https://github.com/inthehand/32feet/issues/61

closed 2020-02-13 by peterfoot