inthehand/32feet #54: BluetoothRadio.PrimaryRadio return null?

lhappyb, 2019-08-01:
Windows 7 64bit，VS2010，use InTheHand.Net.Personal.dll
bluetooth device : CSR Bluetooth Radio

`BluetoothRadio myRadio = BluetoothRadio.PrimaryRadio;`

the myRadio return null?

API Document:
For Windows CE based devices this is the only BluetoothRadio, for Windows XP this is the first available BluetoothRadio device.
If the device has a third-party stack this property will return null

**is the CSR Bluetooth Radio is third-party stack ?**

`BluetoothClient client = new BluetoothClient();` also is error

DiscoverDevices API can't be run。

please help to find what error ，thanks

closed 2020-08-07 by peterfoot

Ikrom27, 2022-02-10:
How did you fix it?

peterfoot, 2022-02-10:
The documentation needs updating as since Windows 7 the third-party stack comment doesn't really apply - there are add-ins like Broadcom which add profiles but work with the Microsoft stack. The most common reason for BluetoothRadio or BluetoothClient failing like this is using the "empty" .NET Standard dll rather than the Win32 implementation. When you add the NuGet to the executable project this is handled automatically for you.