inthehand/32feet #59: The capabilities of 32feet stack - is the FTP server role supported?

loredana1989, 2019-08-13:
Hi,
 I'm trying to develop a Bluetooth Windows form application in c# and I'm using 32feet library (3.5.0).
Currently I'm struggling with FTP profile (SPP profile is working great). 

Could someone please tell me if FTP server role is supported by 32feet stack? I'm struggling for a few days and I couldn't find any answer for my issue. 

I would be very grateful for an answer.
Loredana

peterfoot, 2019-08-13:
There isn't built in functionality but you can use 32feet to implement one. The FTP server service is implemented over RFComm using OBEX. You'd need to read the spec and implement the server code using the BluetoothListener or a listening socket directly.

loredana1989, 2019-08-14:
Thanks for your response. 
If you already have some documentation please send to me. 
I've tried:
 
BluetoothListener bluetooth = new BluetoothListener(BluetoothService.ObexFileTransfer);
bluetooth.Start();
bluetooth.AcceptBluetoothClient();

and is not working.