inthehand/32feet #64: relative path not found

mtyeager, 2020-01-20:
Sorry, probably a dumb question, but I loaded the solution into VS2019, and it is failing to build in part due to two references to IOBluetooth.framework and IOBluetoothUI.framework. Wondering what those are and how best to resolve. See picture.
[image]

peterfoot, 2020-01-20:
Thanks for spotting this, I will need to fix these to paths which work regardless of the project file location. The \System\Library\Frameworks folder should be present on all macs with XCode installed so should work as a fixed absolute path in this case.

mtyeager, 2020-01-20:
Hi Peter,

I am working using VS on Windows 10. Was the intention that this be workable on Windows? Or only Mac?

Thanks,

Michael

peterfoot, 2020-01-20:
The native macOS libraries should only build on a Mac. Temporarily you can exclude them from the build but I need to tweak the solution to work around this.

Peter

closed 2023-10-11 by peterfoot