inthehand/32feet #65: Silicon labs Bluetooth Peripheral  device not discovered using 32Feet.Net library

alpesh-einfochips, 2020-02-12:
I am trying to discovered Silicon labs Bluetooth Peripheral device , but not discovered with 32Feet.Net library API, and we can discovered other Android bluetooth devices.
I test with Window 10 system bluetooth setting and we can found silicon labs device also.

32Feet.Net with discovered silicon labs bluetooth device only  when system window Bluetooth is discovering  working.

My code:
// mac is mac address of local bluetooth device
BluetoothEndPoint localEndpoint = new BluetoothEndPoint(BluetoothRadio.PrimaryRadio.LocalAddress, BluetoothService.BluetoothBase);
// client is used to manage connections
BluetoothClient localClient = new BluetoothClient(localEndpoint);
// component is used to manage device discovery
BluetoothComponent localComponent = new BluetoothComponent(localClient);
// async methods, can be done synchronously too
localComponent.DiscoverDevicesAsync(255, false, false, true, false, null);
 localComponent.DiscoverDevicesProgress += new EventHandler<DiscoverDevicesEventArgs>(component_DiscoverDevicesProgress);
localComponent.DiscoverDevicesComplete += new EventHandler<DiscoverDevicesEventArgs>(component_DiscoverDevicesComplete);](url)

private void component_DiscoverDevicesProgress(object sender, DiscoverDevicesEventArgs e)
{
            isSearchingDevices = true;
            // log and save all found devices
            for (int i = 0; i < e.Devices.Length; i++)
            {
                if (e.Devices[i].Remembered)
                {
                    Console.WriteLine(e.Devices[i].DeviceName + " (" + e.Devices[i].DeviceAddress + "): Device is known");
                }
                else
                {
                    Console.WriteLine(e.Devices[i].DeviceName + " (" + e.Devices[i].DeviceAddress + "): Device is unknown");
                }
                if (e.Devices[i].DeviceName == "Empty Example")
                {
                    Console.WriteLine(e.Devices[i].DeviceName) ;
                }
                devicesList.Add(new Device(e.Devices[i]));
            }
 }

private void component_DiscoverDevicesComplete(object sender, DiscoverDevicesEventArgs e)
{
    isSearchingDevices = false;
}

alpesh-einfochips, 2020-02-13:
We discovered but not paring, please help me out
Bluetooth android device with Pairing but not with BLE device:
My pairing code:

        public void pairDevice(Device device)
        {
            {
                // if the device is not paired, pair it!
                {
                    EventHandler<BluetoothWin32AuthenticationEventArgs> handler = new EventHandler<BluetoothWin32AuthenticationEventArgs>(HandleRequests);
                    BluetoothWin32Authentication auth = new BluetoothWin32Authentication(handler);
                    // replace DEVICE_PIN here, synchronous method, but fast
                    bool isPaired = BluetoothSecurity.PairRequest(device.DeviceInfo.DeviceAddress, null);
                    if (isPaired)
                    {
                        // now it is paired
                        ShowMessage("Device is paired successfully");

                        isSearchingDevices = false;
                        discoveryDevices();
                    }
                    else
                    {
                        // pairing failed
                        ShowMessage("Device pairing fail");

                    }
                    auth.Dispose();
                }
            }
        }
private void HandleRequests(object that, BluetoothWin32AuthenticationEventArgs e)
        {
            e.Confirm = true;
        }

peterfoot, 2020-02-13:
The InTheHand.Net.Bluetooth API is for Bluetooth Classic only. We're working on a separate dll for Bluetooth LE which will be available soon.

closed 2020-02-13 by peterfoot