inthehand/32feet #66: .Net core 2.1 and .Net standard 2.0

pirhano, 2020-02-25:
The library runs fine in .Framework environment only.
When its used on .Net Core no devices are detected same as .Net Standard

Slipoch, 2020-03-11:
+1 to this. DiscoverDevices not working on .net std 2.1, 2.0 or .net core 3.1.

atsgeiger, 2020-04-29:
I think have seen the same thing, using latest stable from NuGet, 4.0.4 as well as 4.0.2

I have a Class Library set to target .NET Standard 2.0, and code within that library:

```BluetoothClient client = new BluetoothClient();
IReadOnlyCollection<BluetoothDeviceInfo> devices = client.DiscoverDevices(50);```

When I reference this Class Library from a Windows Forms targeting .NET Framework 4.6, I instantiate the the class and call the code that runs the BluetoothClient code you see above, it finds all of my devices in about 5 seconds or so.

When I reference this Class Library from a UWP (target win10 1903), Mac (??), Android (target v9), or Xamarin.Forms (target .net standard 2.0) projects it does not work at all. Am I missing something stupid?

peterfoot, 2020-04-30:
At the moment not all the platforms are implemented. The .NET Framework library works on Windows Desktop projects only. UWP requires a separate implementation because the Win32 APIs used are not permitted inside a UWP app.
The .NET Standard dll is a reference library with no implementation. The relevant platform dll will be deployed into the end application when you reference the NuGet library.  I'll add a matrix of platform support to the documentation to show how the library is progressing.

peterfoot, 2020-07-03:
I'd welcome any contributions to the library. I'm hopeful that with the architectural changes in v4 that there wouldn't be significant work to also build a .NET Core version of the dll. The problem as I see it is ensure there are runtime specific versions as .NET Core can run across Windows, Linux etc and these require different implementations. I see this as an important piece as we move towards .NET 5 and conversion of the different runtimes. The table in the Wiki does indeed show planned support so while some of the underlying macOS support is there it is not currently supported in InTheHand.Net.Bluetooth.

peterfoot, 2021-12-21:
.NET Core 3.1 and .NET 5.0 support added

closed 2021-12-21 by peterfoot