inthehand/32feet #67: can not receive data from mobile

willhuo, 2020-03-10:
win10 enterprise  
OS version 1803

I wrote a winfrom program. And it can discover mobile bluetooth and can connect it.
But i can not receieve data or send data via the stream object.

My sample code as below.

` public partial class Home : Form
    {
        public Home()
        {
            InitializeComponent();
        }
        private void Home_Load(object sender, EventArgs e)
        {
            Utility.InMemorySink.OnLogReceivedEvent += InMemorySink_OnLogReceivedEvent;
            ScanBluetooth();
        }

        private BluetoothClient _Blueclient = new BluetoothClient();
        private Dictionary<string, BluetoothDeviceInfo> _DeviceInfos = new Dictionary<string, BluetoothDeviceInfo>();
        private Queue<string> _SendStrQueue = new Queue<string>();

        private void btnScan_Click(object sender, EventArgs e)
        {
            ScanBluetooth();
        }
        private void btnPrintDevice_Click(object sender, EventArgs e)
        {
            string deviceName = comBluetooths.Text;
            BluetoothDeviceInfo deviceInfo = _DeviceInfos[deviceName];
            Serilog.Log.Information("device name：{0}， address：{1}，state：{2}，ClassOfDevice：{3}，Authenticated：{4}，Remembered：{5}，LastSeen：{6}，LastUsed：{7}, Connected:{8}", 
                deviceInfo.DeviceName,
                deviceInfo.DeviceAddress.ToString(),
                deviceInfo.Connected,
                deviceInfo.ClassOfDevice,
                deviceInfo.Authenticated,
                deviceInfo.Remembered,
                deviceInfo.LastSeen,
                deviceInfo.LastUsed,
                _Blueclient.Connected);
        }
        private void btnConnect_Click(object sender, EventArgs e)
        {
            Connect();
        }
        private void btnSendData_Click(object sender, EventArgs e)
        {
            var str = txtSendData.Text.Trim();
            if(string.IsNullOrEmpty(str))
            {
                Serilog.Log.Warning("send data is null or empty");
                return;
            }

            str += Environment.NewLine;
            _SendStrQueue.Enqueue(str);
        }

        private void ScanBluetooth()
        {
            var ra= BluetoothRadio.AllRadios;
            BluetoothRadio bluetoothRadio = BluetoothRadio.PrimaryRadio;
            if (bluetoothRadio == null)
                Serilog.Log.Warning("can not find bluetooth device");
            bluetoothRadio.Mode = RadioMode.Connectable;
            BluetoothDeviceInfo[] Devices = _Blueclient.DiscoverDevices();
            _DeviceInfos.Clear();
            comBluetooths.Items.Clear();
            foreach (BluetoothDeviceInfo device in Devices)
            {
                comBluetooths.Items.Add(device.DeviceName);
                _DeviceInfos[device.DeviceName] = device;
            }

            Serilog.Log.Information("ClassOfDevice: {0}" , bluetoothRadio.ClassOfDevice);
            Serilog.Log.Information("HardwareStatus: {0}" , bluetoothRadio.HardwareStatus);
            Serilog.Log.Information("HciRevision: {0}" , bluetoothRadio.HciRevision);
            Serilog.Log.Information("HciVersion: {0}" , bluetoothRadio.HciVersion);
            Serilog.Log.Information("LmpSubversion: {0}" , bluetoothRadio.LmpSubversion);
            Serilog.Log.Information("LmpVersion: {0}" , bluetoothRadio.LmpVersion);
            Serilog.Log.Information("LocalAddress: {0}" , bluetoothRadio.LocalAddress);
            Serilog.Log.Information("Manufacturer: {0}" , bluetoothRadio.Manufacturer);
            Serilog.Log.Information("Mode: {0}" , bluetoothRadio.Mode);
            Serilog.Log.Information("Name: {0}" , bluetoothRadio.Name);
            Serilog.Log.Information("Remote: {0}" , bluetoothRadio.Remote);
            Serilog.Log.Information("SoftwareManufacturer: {0}" , bluetoothRadio.SoftwareManufacturer);
            Serilog.Log.Information("StackFactory: {0}" , bluetoothRadio.StackFactory);
            Serilog.Log.Information("***********************************");
        }
        private void Connect()
        {
            try
            {              
                string deviceName = comBluetooths.Text;
                BluetoothDeviceInfo deviceInfo = _DeviceInfos[deviceName];

                //_Blueclient.SetPin(deviceInfo.DeviceAddress, "000000");   
                _Blueclient.Connect(deviceInfo.DeviceAddress, BluetoothService.Handsfree);

                if (_Blueclient.Connected)
                {
                    Serilog.Log.Information("connect success");
                }
                else
                {
                    Serilog.Log.Information("connect faild");
                }

                //transfer data.....
                var stream = _Blueclient.GetStream();
                Serilog.Log.Information("get stream success");

                Task.Run(() => RecvData(stream));
                Serilog.Log.Information("begin to listen bluetooth data");
            }
            catch (Exception ex)
            {
                Serilog.Log.Error(ex, "bluetooth open error");
            }
        }
        private void RecvData(Stream stream)
        {
            var flag = true;
            var buff = new byte[100];
            while (flag)
            {
                var count= stream.Read(buff, 0, buff.Length);
                if(count==0)
                {
                    //Serilog.Log.Information("count={1}", count);
                    if (_SendStrQueue.Count > 0)
                    {
                        var sendStr = _SendStrQueue.Dequeue();
                        SendMsg(stream, sendStr);
                        continue;
                    }
                    Task.Delay(100).Wait();
                    continue;
                }
                
                var recvstr = Encoding.ASCII.GetString(buff, 0, count);
                Serilog.Log.Information("recvstr: {0}, count={1}",recvstr,count);

                //parser
                Parse(stream, recvstr);
            }
        }
        private void Parse(Stream stream,string recvstr)
        {
            SendMsg(stream, recvstr);
        }
        private void SendMsg(Stream stream,string msg)
        {
            byte[] sendbuf = Encoding.ASCII.GetBytes(msg);
            stream.Write(sendbuf, 0, sendbuf.Length);
            Serilog.Log.Information("sendstr: {0}", msg);
        }
        private void InMemorySink_OnLogReceivedEvent(object sender, string e)
        {
            richTxtLog.Invoke(new Action(() =>
            {
                richTxtLog.AppendText(e + Environment.NewLine);
                richTxtLog.ScrollToCaret();
            }));
        }

        static EventHandler<BluetoothWin32AuthenticationEventArgs> handler = new EventHandler<BluetoothWin32AuthenticationEventArgs>(handleRequests);
        BluetoothWin32Authentication authenticator = new BluetoothWin32Authentication(handler);

        private static void handleRequests(Object thing, BluetoothWin32AuthenticationEventArgs args)
        {
            args.Confirm = true;
        }
    }`

closed 2020-04-08 by willhuo

peterfoot, 2023-02-23:
it looks from the code above that Stream.Flush is not called - anything you write will be buffered and not actually sent to the remote device until you call Flush on the stream. It's likely this is what is happening above.

willhuo, 2023-03-01:
@fuhongbing134  As peterfoot said,It may be 'stream.flush' lead to the problem.I almost forgot how I handled this problem two years ago.