inthehand/32feet #69: How to handle incoming Bluetooth events in 4.0.2?

aronneberg, 2020-03-31:
Good Morning,

I'm new to this library and a little bit confused with how to handle incoming Bluetooth events like SSP pairing requests initiated on another device. In the documentation I found the BluetoothListener and BluetoothWin32Authentication classes which seem to be intended for this task. However, in one of the recent commits (and all 4.0.x versions available via Nuget), the BluetoothListener class was replaced with partial classes and it eludes me how to use the new structure to respond to BT events. Is there any updated documentation on handling incoming BT events with the current version?

Cheers
Andreas

peterfoot, 2020-03-31:
At the moment v4 contains a subset of v3. BluetoothListener is in development and will be available soon. Better support for Bluetooth pairing will follow later but not in a way which is tied just to the Win32 implementation. For now you can use the v3.5/v3.7 versions to use the full functionality on Win32.

aronneberg, 2020-04-07:
Thanks for your reply, I was able to implement the needed functionality in 3.5.

closed 2020-04-07 by aronneberg

juliokele, 2020-10-02:
> At the moment v4 contains a subset of v3. BluetoothListener is in development and will be available soon. Better support for Bluetooth pairing will follow later but not in a way which is tied just [...]

Any news on this?
I need to respond (confirm) to BT events in v4.