inthehand/32feet #70: Check bluetooth battery level

sturla78, 2020-04-14:
Hi,
I have use this library under windows 10 using the "InTheHand.Net.Sockets" and discovering the device with BluetoothClient class.
All work fine, i can show the device and the property of the BluetoothDeviceInfo. But what i'm try to do, is check the battery level of my headphone device (in the specific a  JBL Tune500BT), i have search on exposed service but i cannot find any property for this scope.

Am I doing something wrong or this is the correct way ? (or the API are'nt capable of this)

Thanks.

peterfoot, 2020-07-03:
This is most likely implemented by the Bluetooth Low Energy Battery service. This will be soon be available in the InTheHand.BluetoothLE library for cross platform use. However you can currently use the Windows.Devices.Bluetooth.GenericAttributeProfile namespace directly on Windows 10.

sturla78, 2020-07-03:
Thank you for the idea, i will try to use the namespace implementation of Windows.Devices.Bluetooth.GenericAttributeProfile.

soyxan, 2020-07-09:
Did you had luck with this?. I am trying to do exactly the same for mi Jabra Elite 25e HeadSet.

I have checked your JBL TUNE 500BT and it seems that they are not using BLE (Bluetooth Low Energy). They are using Bluetooth 4.1 (sames as mines). So it seems that we need to use Bluetooth Classic approach, specifically RFCOMM. I have managed to connect to my headset using RFCOMM sockets, but I am not able to get any response from the headset.

In the following link you have a python code that does that (RFCOMM) for several bluetooth headset to get the battery level, but it will not work in Windows:

https://github.com/TheWeirdDev/Bluetooth_Headset_Battery_Level/blob/master/bluetooth_battery.py

I will be very appreciated if you can share with us your advances regarding this.

Thanks!

sturla78, 2020-08-01:
Hello to everybody,
sorry for the late response, i have try a couple of test and this it took some time.
Unfortunately i haven't find any solution that work for this purpose. As correctly says soyxan i cannot use the bluetooth LE way, in this case if i try to use this type of object with my headset, an error is returned that said that my device id is not correct for the bluetooth LE tecnology.
So i have try another way, using rfcomm and the  BluetoothDevice object exposed by Windows.Devices.Bluetooth namespace, i have build a windows desktop app and a UWP app, but both application with the same result, i can open a stream using GetRfcommServicesAsync methid but nothing happen, the stream is always empty also if i try to send an AT commant. 
What is certain is that is possible retrieve this infomation becouse this app https://www.bluetoothgoodies.com/ is able to find correctly the battery level of my headset, but i don't know how it does.

soyxan, 2020-08-01:
I managed to communicate with my Headset using RFCOMM, check this issue #75

sturla78, 2020-08-19:
Thank you soyxan,
with the  "SetServiceState" method i managed to get a correct comunication with my headset, and using the exchange of AT commands i got  the battery level information.

Thanks again.

steam3d, 2022-12-07:
@sturla78 How do you get battery information? 

I received string:
`AT+IPHONEACCEV=2,1,1,2,0` (Windows shows 30%)
`AT+IPHONEACCEV=2,1,9,2,0` (Windows shows 100%)

Looks like the battery is number 3. But this is at odds with the data from the documentation

[image]

closed 2025-12-14 by peterfoot