inthehand/32feet #72: Discover Devices Async

ske66, 2020-05-27:
I've been looking through the documentation and it mentions the ability to look for devices asynchronously but no such methods seem to be available. The documentation talks about a class called BluetoothComponent with a method called DiscoverDevicesAsync. But I couldn't find this class or method in the most recent version of the library.

Additionally, the Class Library Documentation mentions a BeginDiscoverDevices method with an argument for AsyncCallback, this is part of the BluetoothClient class I couldn't find this method anywhere.

Can you please advise what methods can be used for asynchronous device searching.

peterfoot, 2020-07-03:
I'm currently looking at this as an enhancement. The old v3.x version of the library did introduce a Begin* End* async pattern which fitted with the Win32 Sockets approach at the time but this is no longer recommended and doesn't apply easily to other platforms.

nreh, 2023-07-22:
> The old v3.x version of the library did introduce a Begin* End* async pattern which fitted with the Win32 Sockets approach at the time but this is no longer recommended and doesn't apply easily to other platforms.

Yep, was trying to create a nice UWP UI that automatically showed devices as they were found, but because the old library breaks my project I can't use the old async functionality. Anyone know a workaround?

perkjelsvik, 2025-03-21:
Hi! I was wondering if there is any progress on this issue for having asynchronous discovery on Bluetooth Classic?