inthehand/32feet #76: [Question] Socket exception: The operation completed successfully.

jcs090218, 2020-07-17:
I am trying to connect to my raspberry pi device, yet no luck doing it. I am getting `Socket exception: The operation completed successfully.` message and I have no idea why is an error message saying something successfully? 😕 

Here is the code, what am I doing wrong? Thanks!

```cs
var cli = new BluetoothClient();
IReadOnlyCollection<BluetoothDeviceInfo> peers = cli.DiscoverDevices();

foreach (var bdi in peers)
{
    print(bdi.DeviceName);

    if (bdi.DeviceName == "raspberrypi")
        mDevice = bdi;
}

if (mDevice != null)
    print("Address: " + mDevice.DeviceAddress);

// Error here!
cli.Connect(mDevice.DeviceAddress, BluetoothService.SerialPort);

NetworkStream stream = cli.GetStream();
```

jcs090218, 2020-07-18:
Okay, the same code works in normal C# application but not in Unity. :(

peterfoot, 2020-08-03:
This should be resolved in 4.0.7 which was just published on NuGet

jcs090218, 2020-08-04:
Thanks! I will try this when I have time! 😄

jcs090218, 2020-08-05:
I have tested with version `4.0.8`! Thanks! 😄

closed 2020-08-05 by jcs090218