inthehand/32feet #77: beginConnect gone?

veitk, 2020-07-24:
Hi, 

first: i really like that you are working on the project again. 

I asked how to cancel a bt-connect after one second about 7-8 years ago and now since i improved my coding skills i'm trying to improve my c# app which i'm still using. 

I found some interesting code (MarcFletcher/NetworkComms.Net) examples to timeout a connect but the current version of your btclient doesn't have a beginConnect method anymore....

not an issue but maybe you can give me a hint:

i need to transfer data to 20 devices (moving people) fast. i try to connect to 7 at a time and if all 7 are in range everything works perfectly. if one device is out of range the connect to this device does block the bluetooth radio. timeout is 5 seconds. 
i want to decrease to 1 second, since if the device is in range it will connect in less than 700ms. 

(the device are not discoverable by the way and do not have any bt-le, only bt with serial port)

peterfoot, 2020-07-24:
In order to support Unity and the Mono runtime the socket functions were wrapped directly from the Win32 API so ConnectAsync and Begin/EndConnect are not currently implemented. 

I think a possible workaround would be to create a separate thread with a reference to the Client socket and call Close() on this socket after the desired timeout if it didn't succeed.

Finn77777, 2020-08-20:
> Hi,
>
>
>
> first: i really like that you are working on the project again.
>
>
>
> I asked how to cancel a bt-connect after one second about 7-8 years ago and now since i improved my coding skills i'm [...]

peterfoot, 2020-08-20:
That's a good suggestion. I would definitely like to add back Async methods however currently the Win32 implementation uses native Winsock functions rather than the .NET implementation of Socket because this is the only working option when running on Mono (Unity etc). Also there isn't an equivalent on Android so would need to find a way to implement in a way which can work across platforms.

peterfoot, 2020-10-29:
4.0.12 adds these back to the Win32 specific version of the assembly. They won't appear in a cross platform context.

closed 2020-10-29 by peterfoot