inthehand/32feet #87: BluetoothSecurity.PairRequest issue in InTheHand.Net.Bluetooth V4

juliokele, 2020-10-05:
After BluetoothSecurity.PairRequest with result=true, paired device in Windows10 Bluetoothmanager not shown.
Aftert that: BluetoothSecurity.RemoveDevice result = false.
Pairing with BluetoothAuthenticationMethod.NumericComparison, NumericComparison on target device not shown.

juliokele, 2020-10-15:
any news on this?

peterfoot, 2020-10-15:
For NumericComparison the handler should be echoing back whatever code was received and so automatically approving it. I'll need to run some more tests. What type of device are you pairing with?

juliokele, 2020-10-15:
in V3 mostly works, but not always, process hangs infinitely at this point:
https://github.com/inthehand/32feet/blob/a3d464e7825f7ed9b458bdcc5755c2fe6bf2ea00/Legacy/InTheHand.Net.Personal/Net.Bluetooth.Msft/WindowsBluetoothSecurity.cs#L282

Thats why i switched to V4...

Device is a custom device with ESP32 Chip on + SPP_INITIATOR_DEMO firmware.
Pairing with Windows 10 (Version: 2004, Build: 19041.572) built in Bluetooth Manager works without any problems

JkPersson, 2022-08-22:
I am seeing the same on Android (12) and MAUI.

`        IReadOnlyCollection<BluetoothDeviceInfo> bluetoothDeviceInfos = bluetoothClient.DiscoverDevices();
        BluetoothDeviceInfo frameworkDevice = bluetoothDeviceInfos.FirstOrDefault(x => x.DeviceAddress.ToString("C") == MACAddress);
        if (frameworkDevice == null)
        {
            await Shell.Current.DisplayAlert("Pair", "Device is not available", AppResources.Close);
            return;
        }
        bool paired = true;
        if (!frameworkDevice.Authenticated)
        {
            paired = BluetoothSecurity.PairRequest(frameworkDevice.DeviceAddress, Pin);
        }
`

PairRequest return true, but the device is not paired. 
Same call with Pin=null does not trigger the Pairing Pin dialog.
I have the following Bluetooth permission set for the app:

	<uses-permission android:name="android.permission.BLUETOOTH" />
	<uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
	<uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
	<uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />

Pyroa, 2023-01-01:
Same issue here !

With MAUI .Net 6.0 and Android 12 SKQ1.210908.001 on MUI 13.0.11
Trying to pair a HC-06 device.

It's working to pairing the device manually.