inthehand/32feet #89: InTheHand.BluetoothLE (4.0.13-pre) - Bluetooth.RequestDeviceAsync -> Invalid Window Handle Exception

juliokele, 2020-10-12:
i got Invalid Window Handle Exception after this (tested in Console Project, TargetFramework: net462):
https://github.com/inthehand/32feet/blob/cb1b8f8f3e991be11ad348d55061cdfe5757b145/Samples/BluetoothClientApp/BluetoothClientApp/MainPage.xaml.cs#L42

Exception:
```
System.Exception: Invalid Window Handle (Exception HRESULT: 0x80070578)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at InTheHand.Bluetooth.Bluetooth.<PlatformRequestDevice>d__17.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
   at BluetoothTest.Program.<Main>d__1.MoveNext() in ...
```

peterfoot, 2020-10-12:
Looks like the Windows 10 picker requires a main window handle when used from a desktop app. Since there is no way to manually pass a window handle to the API there is no easy fix for this. You could use the GetPairedDevicesAsync() method to retrieve a list of paired devices which you can display in the console. I'm working on an API for discovery which returns a list of devices rather than the picker but it's not currently enabled in release builds.

juliokele, 2020-10-12:
I use meantime ScanForDevicesAsync(). Thanks anyway.

My other problem is:
`var rssi = await device.Gatt.ReadRssi()`
ReadRssi() return is always 0  (ReadRssi() is not fully implemented yet)

Do you have a plan to fully implement this (in the Near Future)?

peterfoot, 2020-10-12:
Windows doesn't currently have an equivalent API to get the RSSI for an open connection (it can return RSSI from advertisements). This property is implemented for Android and iOS though.