inthehand/32feet #92: xamarin forms, the connect event returns nothing

DesCreategaDavid, 2020-11-03:
Hello good day, I am implementing the library with the example that comes as a sample for xamarin forms, in line 52 where the break is found, apparently nothing returns and the application ends, I do not know that I could be missing to connect with another device.

[image]

peterfoot, 2020-11-03:
Can you try wrapping the ConnectAsync with a try/catch and see what exception is thrown

DesCreategaDavid, 2020-11-03:
Hello, good morning, I put the try catch and it sent me to this screen, telling me that there is no compatible code running.

[image]

After that mark status 133, in the onClientConnectionState () event.

[image]

TsplayerT, 2021-06-16:
There are no more details about this issue? It also happens to me.

peterfoot, 2021-06-16:
Unfortunately the native error code 133 is GATT_ERROR which is a generic error which can represent a number of issues. What is the type of device you are trying to connect to? Can you access it with a tool like nRF Connect?