inthehand/32feet #93: DllNotFoundException: Unable to load DLL 'bthprops.cpl': The specified module could not be found.

DiemerCohenStuart, 2020-11-06:
Hi @peterfoot ,

Thank you for your reply earlier today. The code works great now in the Unity Editor.
However, after deploying to the Hololens, we are getting the error "DllNotFoundException: Unable to load DLL 'bthprops.cpl': The specified module could not be found."

I suppose the reason is that the bthprops.cpl file is located in Windows/System32, whereas the Hololens runs on ARM (SysArm32). Do you have maybe an idea how to solve this?

Thank you very much again for your help. 
Kind regards, Diemer

peterfoot, 2020-11-06:
Ah, so the Hololens doesn't have all the Win32 APIs, the solution will be to wrap the UWP APIs. This is work that I have just started but is somewhere off completion at the moment.

DiemerCohenStuart, 2020-11-06:
Great! Thanks Peter. Any idea when you expect to finish the work (approximately)? Weeks? Months? 
Otherwise I will try to look for a temporary solution. Thanks again and have a great day, Diemer

peterfoot, 2020-12-11:
I'm going to work on it in the new year, so Months rather than weeks sorry.

DiemerCohenStuart, 2020-12-14:
No worries! Thanks for your reply.

azlogs, 2021-04-29:
> Otherwise I will try to look for a temporary solution. Thanks again and have a great day, Diemer

HI Diemer,
Have u has a temporary solution for this issue? I have the same issue

Thanks,
Phuc Pham

peterfoot, 2021-04-30:
Hi, there's no temporary solution unfortunately, the desktop version of the dll is calling APIs which don't existing in Hololens. There is a partial implementation of the UWP platform in the source but this needs to be completed in order to support Hololens.