inthehand/32feet #94: Calling BluetoothClient Close() causes StackOverflowException

CodeMinion, 2020-11-16:
Hi @peterfoot,

Thanks for building and maintaining this library all these years. 

I may have stumbled onto a defect, but my use case is a bit unique so this may not be specific to the library. I am using the version 4.0.12 of InTheHand.Net.Bluetooth in Unity running on a Windows 8 machine. 

I can establish a Bluetooth connection to the remote device using the BluetoothClient class from Unity just fine, but when I try to call Close() on my client instance I am getting a StackOverflowException. 

`
StackOverflowException: The requested operation caused a stack overflow.
System.Net.Sockets.Socket.Close () (at <aa976c2104104b7ca9e1785715722c9d>:0)
InTheHand.Net.Sockets.Win32Socket.Close () (at <b212d04f28324d63b471489e1555fb35>:0)
InTheHand.Net.Sockets.Win32Socket.Dispose (System.Boolean disposing) (at <b212d04f28324d63b471489e1555fb35>:0)
System.Net.Sockets.Socket.Dispose () (at <aa976c2104104b7ca9e1785715722c9d>:0)
System.Net.Sockets.Socket.Close () (at <aa976c2104104b7ca9e1785715722c9d>:0)
InTheHand.Net.Sockets.Win32Socket.Close () (at <b212d04f28324d63b471489e1555fb35>:0)
InTheHand.Net.Sockets.Win32Socket.Dispose (System.Boolean disposing) (at <b212d04f28324d63b471489e1555fb35>:0)
System.Net.Sockets.Socket.Dispose () (at <aa976c2104104b7ca9e1785715722c9d>:0)
System.Net.Sockets.Socket.Close () (at <aa976c2104104b7ca9e1785715722c9d>:0)
InTheHand.Net.Sockets.Win32Socket.Close () (at <b212d04f28324d63b471489e1555fb35>:0)
InTheHand.Net.Sockets.Win32Socket.Dispose (System.Boolean disposing) (at <b212d04f28324d63b471489e1555fb35>:0)
System.Net.Sockets.Socket.Dispose () (at <aa976c2104104b7ca9e1785715722c9d>:0)
System.Net.Sockets.Socket.Close () (at <aa976c2104104b7ca9e1785715722c9d>:0)
InTheHand.Net.Sockets.Win32Socket.Close () (at <b212d04f28324d63b471489e1555fb35>:0)
InTheHand.Net.Sockets.Win32Socket.Dispose (System.Boolean disposing) (at <b212d04f28324d63b471489e1555fb35>:0)
System.Net.Sockets.Socket.Dispose () (at <aa976c2104104b7ca9e1785715722c9d>:0)
System.Net.Sockets.Socket.Close () (at <aa976c2104104b7ca9e1785715722c9d>:0)
InTheHand.Net.Sockets.Win32Socket.Close () (at <b212d04f28324d63b471489e1555fb35>:0)
InTheHand.Net.Sockets.Win32Socket.Dispose (System.Boolean disposing) (at <b212d04f28324d63b471489e1555fb35>:0)
System.Net.Sockets.Socket.Dispose () (at <aa976c2104104b7ca9e1785715722c9d>:0)
System.Net.Sockets.Socket.Close () (at <aa976c2104104b7ca9e1785715722c9d>:0)
InTheHand.Net.Sockets.Win32Socket.Close () (at <b212d04f28324d63b471489e1555fb35>:0)
InTheHand.Net.Sockets.Win32Socket.Dispose (System.Boolean disposing) (at <b212d04f28324d63b471489e1555fb35>:0)
System.Net.Sockets.Socket.Dispose () (at <aa976c2104104b7ca9e1785715722c9d>:0)
System.Net.Sockets.Socket.Close () (at <aa976c2104104b7ca9e1785715722c9d>:0)
InTheHand.Net.Sockets.Win32Socket.Close () (at <b212d04f28324d63b471489e1555fb35>:0)
InTheHand.Net.Sockets.Win32Socket.Dispose (System.Boolean disposing) (at <b212d04f28324d63b471489e1555fb35>:0)
System.Net.Sockets.Socket.Dispose () (at <aa976c2104104b7ca9e1785715722c9d>:0)
System.Net.Sockets.Socket.Close () (at <aa976c2104104b7ca9e1785715722c9d>:0)
InTheHand.Net.Sockets.Win32Socket.Close () (at <b212d04f28324d63b471489e1555fb35>:0)
InTheHand.Net.Sockets.Win32Socket.Dispose (System.Boolean disposing) (at <b212d04f28324d63b471489e1555fb35>:0)
System.Net.Sockets.Socket.Dispose () (at <aa976c2104104b7ca9e1785715722c9d>:0)
System.Net.Sockets.Socket.Close () (at <aa976c2104104b7ca9e1785715722c9d>:0)
InTheHand.Net.Sockets.Win32Socket.Close () (at <b212d04f28324d63b471489e1555fb35>:0)
InTheHand.Net.Sockets.Win32Socket.Dispose (System.Boolean disposing) (at <b212d04f28324d63b471489e1555fb35>:0)
System.Net.Sockets.Socket.Dispose () (at <aa976c2104104b7ca9e1785715722c9d>:0)
System.Net.Sockets.Socket.Close () (at <aa976c2104104b7ca9e1785715722c9d>:0)
InTheHand.Net.Sockets.Win32Socket.Close () (at <b212d04f28324d63b471489e1555fb35>:0)
InTheHand.Net.Sockets.Win32Socket.Dispose (System.Boolean disposing) (at <b212d04f28324d63b471489e1555fb35>:0)
System.Net.Sockets.Socket.Dispose () (at <aa976c2104104b7ca9e1785715722c9d>:0)`

Thank you once again,

peterfoot, 2020-11-16:
I've verified this and am working on a fix.

CodeMinion, 2020-11-16:
Thanks @peterfoot !

closed 2021-12-21 by peterfoot