inthehand/32feet #96: AdvertisementReceived event always gives e.Device null

aderregibus, 2020-11-19:
Code below throws an exception because e.Device is null.
Scan works and advertisement event is rised when MY_DEVICE_NAME is found, but I'm stuck there...  can this be a bug?

        private async void ScanButton_Click(object sender, EventArgs e)
        {
            Bluetooth.AdvertisementReceived += Bluetooth_AdvertisementReceived;
            bluetoothLEScan = await Bluetooth.RequestLEScanAsync();
        }

        private async void Bluetooth_AdvertisementReceived(object sender, BluetoothAdvertisingEvent e)
        {
            Console.WriteLine($"Name:{e.Name} Rssi:{e.Rssi}");

            if (e.Name == "MY_DEVICE_NAME")
            {
                await e.Device.Gatt.ConnectAsync();
                var servs = await e.Device.Gatt.GetPrimaryServicesAsync();
            }
        }

aderregibus, 2020-11-24:
Anyone experiencing this same issue?

I'm using .NET 4.6.1 on Windows 10. If any additional information is required, please advise.

peterfoot, 2020-11-24:
Hi, I've looked at this with a couple of devices and it does return a BluetoothDevice however it may depend on the advertising device and the type of Bluetooth address in the native advertisement. If it's Unspecified then no address is exposed and if it is Random then Windows may not return a device object from the address. I need to see if in some cases there should be a "fake" BluetoothDevice which contains the random device address. What kind of device are you watching advertisements from?

aderregibus, 2020-11-24:
Thanks for your quick reply.
I'm testing using two different BLE devices. One is a custom product I'm developing over an ESP32, advertising with public address, and the other is a temperature sensor.
Both cases are returning null device (and also some unknown devices that are picked up from the surrounding also return null device).
Additionally, I've downloaded IPWorks BLE demo and sucessfully connected to both devices on the same Windows machine and .NET framework. I know they are different things, but just to add information in case you find it useful to diagnose.
Thank you.

workbgm, 2021-04-13:
I have the same problem, I hope it can be solved

workbgm, 2021-04-13:
device :lairpods

workbgm, 2021-04-13:
device :airpods

workbgm, 2021-04-14:
I find that if I use“ Windows.Devices.Bluetooth ”The Bluetooth address is random too.
`  private static BluetoothLEAdvertisementWatcher bluetoothLEAdvertisementWatcher;

        public static void Read()
        {
            bluetoothLEAdvertisementWatcher = new BluetoothLEAdvertisementWatcher();
            bluetoothLEAdvertisementWatcher.Received += OnAdvertisementReceived;
            bluetoothLEAdvertisementWatcher.Start();

        }

        private static async void OnAdvertisementReceived(BluetoothLEAdvertisementWatcher watcher, BluetoothLEAdvertisementReceivedEventArgs eventArgs)
        {

            var manufacturerSections = eventArgs.Advertisement.ManufacturerData;
            
            if (manufacturerSections.Count > 0 )
            {
                var manufacturerData = manufacturerSections[0];
                if (manufacturerData.CompanyId == 76)
                {
                    if (manufacturerData.Data.Length == 27)
                    {
                        var data = new byte[manufacturerData.Data.Length];
                        using (var reader = DataReader.FromBuffer(manufacturerData.Data))
                        {
                            reader.ReadBytes(data);
                            Console.WriteLine(eventArgs.BluetoothAddress.ToString("x8"));
                            parseData(byteToHexStr(data));
                        }
                    }
                }
              
                
            }
        }`
Ask a question about Bluetooth protocol. If it's random, how can I confirm who sent it？

workbgm, 2021-04-14:
If you use its Bluetooth library on Android, it becomes very simple.
`  @Override
                public void onScanResult (int callbackType, ScanResult result) {
                    try {
                        byte[] data = Objects.requireNonNull(result.getScanRecord()).getManufacturerSpecificData(76);`
It's easy for me to know which device send "manufacturer specific data"

closed 2023-10-11 by peterfoot

chukwuifeanyi, 2024-11-26:
It's 2024 and this issue still exists on my windows WPF project using a custom IOT Bluetooth project. Works fine for other non-windows implementation. Are we getting a fix?

chukwuifeanyi, 2024-11-29:
A quick response to this inquiry would be really kind.
Thank you