inthehand/32feet #98: refresh the stream without new client instance?

LeiYangGH, 2020-11-25:
Hi this is really the best ble lib i can found in github.
i am able to use the lib to read my eletronic scale/balance weight.
the code is here: https://github.com/souws/dotnet-bluetooth-weight

first i `new BluetoothClient()`, then `DiscoverDevices()`, then choose my device by name, then `client.Connect(foundD.DeviceAddress, BluetoothService.SerialPort)` here i'm not sure `SerialPort` is the correct option, 
but i got the correct value(same weight as show in the device).

the problem is when the real weight changes, i try to read again via 'client.GetStream().ReadByte()', i always got the same old value(same as when first connect).

if i create another new client instance, then connect, then read again, i can get the new value. but it is much slower, and i think there must be some better ways.

is there any similiar event like `SerialPort`'s `DataRecieved`?

peterfoot, 2020-11-25:
It depends on how the scale reports its values. Many send back continuous readings and so you'll need to use a loop to keep reading from the stream. Then you should see changes over time. Keep the client open throughout the connection and keep a single reference to the stream.

LeiYangGH, 2020-11-26:
thanks,
i got one bluetooth dubgger and got to know the device sends weights at very high frequency:
such as

11 11 11 11 11 21 22 22 33 33...

and i only tried several times before i reached the changed value...

the problem is solved. thanks!

closed 2021-02-01 by LeiYangGH