inthehand/32feet #99: BluetoothListener/BluetoothClient behavior changed by 2c2cd6

tylersouthard, 2020-12-02:
Commit 2c2cd6ea35346a8d7456328820ae0ff381ac7d78 changed BluetoothClient.win32.cs to hold a Socket instead of a Win32Socket. This change causes the code below to function differently (broken without changes) than before.

`
ServiceRecordBuilder builder = new ServiceRecordBuilder();
builder.AddServiceClass(BluetoothService.MessageNotificationServer);
builder.ProtocolType = BluetoothProtocolDescriptorType.GeneralObex;
builder.AddBluetoothProfileDescriptor(BluetoothService.MessageAccessProfile, 1, 4);
builder.ServiceName = "Messaging Notifications";
builder.AddCustomAttribute(InTheHand.Net.Bluetooth.AttributeIds.HandsFreeProfileAttributeId.SupportedFeatures, ElementType.UInt16, 4);
var record = builder.ServiceRecord;

BluetoothListener mnsServerListener = new BluetoothListener(BluetoothService.MessageNotificationServer, record)
{
       ServiceName = "MAP-MNS Windows"
};
mnsServerListener.Start();

var client = mnsServerListener.AcceptBluetoothClient();
var notificationStream = client.GetStream();
`

The key difference is that now the call to client.Connected returns false instead of true, since it is referencing (Socket)Connected instead of (Win32Socket)Connected. This results in client.GetStream() returning null.

Environment:
- InTheHand.Net.Bluetooth 4.0.12
- .NET Framework 4.7.2 app running on Win10 v2004

peterfoot, 2020-12-11:
This was due to a change in the use of Socket or Win32Socket. The latter is now used only for Mono and .NET will use the regular .NET Socket. Fixed in 4.0.13

closed 2020-12-11 by peterfoot

tylersouthard, 2020-12-15:
@peterfoot Thanks for addressing the issue. I'm finding that using the above code (along with a call to register for text message notifications) isn't working when referencing the nuget, but it does when directly referencing the project (at 1d4a72). I tried this in a couple different projects with the same result. When using the nuget, the dll being copied to the executable directory seems to be the right one (has the right version and git sha in the details). I'm wondering if you have any thoughts as to why this might be the case?