inthehand/32feet #102: BluetoothLE.ScanForDevicesAsync is too slow it always takes 30 sec for discover

polipo, 2021-01-22:
I'm use the following code for discover a device with BluetoothLE,
is there a solution?
Question : there's anther way (faster) than scan, a direct opening by macaddress?
I have also tried the AdvertisementReceived event but RequestLEScanAsync it end immediatly without fires the event

 //SCAN CODE THIS TAKES 30 SECOND FOR FOUND ANY DEVICE

            RequestDeviceOptions ro = new RequestDeviceOptions();
            ro.AcceptAllDevices = true;

            DateTime dtStart = DateTime.Now;
            var t = Bluetooth.ScanForDevicesAsync(ro);
            while (t.IsCompleted == false)
            {
                Thread.Sleep(100);
            }
            var ms = (DateTime.Now - dtStart).TotalSeconds; //always 30 sec

peterfoot, 2021-01-22:
If you have previously scanned for a device you can save the Id (and opaque string the nature of which varies by platform but uniquely identifies a device). Then use BluetoothDevice.FromIdAsync to get that device in a new session without doing a full scan.

ITemplate, 2023-05-05:
@peterfoot just tried to start a new process of my application, then call BluetoothDevice.FromIdAsync with a saved opaque Id. I do get a BluetoothDevice alright, but all attempts to either connect or read service fails or returns null. Any pointers? Or maybe the past couple of years there are other ways to scan for a device (that does not take 30 sec)? Thanks.

igiona, 2024-02-04:
> @peterfoot just tried to start a new process of my application, then call BluetoothDevice.FromIdAsync with a saved opaque Id. I do get a BluetoothDevice alright, but all attempts to either connect [...]

I'm facing the same issue here.
`BluetoothDevice.FromIdAsync` works on iOS but doesn't on Android.
The phone is bonded to the device, but still it can't reconnect to the device which is retrieved from that function.

@peterfoot is there anything that you suggest debugging? I'm happy to help you out here 👍