inthehand/32feet #103: Regression in 4.0.14 breaks long working Android Bluetooth Classic streamReader.Read

thhowl, 2021-01-25:
I did not update from 4.0.8 to 4.0.14 until just today.  

Android Bluetooth Classic data streaming was broken by the Oct 29, 2020 commit that changed NetworkStream : Stream  to  NonSocketNetworkStream : NetworkStream, causing StreamReader.Read to now never return.  

The following basic code is an example of what worked in 4.0.8 but in 4.0.14 no longer returns from the call to streamReader.Read:

`BluetoothClient client = new BluetoothClient();`
`client.Connect(someAddress, BluetoothService.SerialPort);`
`System.IO.Stream stream = client.GetStream();`  (see NOTE below regarding type of stream)
`System.IO.StreamReader streamReader = new StreamReader(stream, Encoding.ASCII);`
`char[] streamReaderCharBuffer;`
`int charsRead = streamReader.Read(streamReaderCharBuffer, index: 0, bufferSize: 250);`  <-- this call never returns
`string text = new string(streamReaderCharBuffer, startIndex: 0, charsRead);`

NOTE: Whether one uses stream of type  InTheHand.Net.Sockets.NonNetworkStream, or System.Net.Sockets.NetworkStream, or System.IO.Stream,  the above code still results in a call that never returns.

Sorry I do not have time to investigate more to provide more info, but I assure you this is not related to buffer sizer or my server, but that simply updating from 4.0.8  to 4.0.14  broke the above simple sequence, and reverting to 4.0.8 restores my Xamarin.Forms app’s functionality.

thhowl, 2021-01-25:
The breaking change was made in 4.0.12.

Zahir-Khan, 2021-07-02:
Facing this problem in  4.0.18. Setting the `Stream.ReadTimeout` seems to have no effect on this problem. It seems to be stuck for ever. Noticed that when this happens the write function does not send the bluetooth data also. Works one time then fails. ie one write and read. 

 I wish 4.0.8 worked on latest stable xamarin forms. Let me know if someone knows the version combinations that work.

Zahir-Khan, 2021-07-03:
Update to this. Looks like it gets stuck when the Bluetooth reciever does not send any Packers back. When it does send something read works.

thhowl, 2021-07-03:
Version 4.0.8 works with latest Xamarin.Forms 5.  

I tried 4.0.17 a few weeks back, and then 4.0.12, and noted "every version since 4.0.12 crashes in PlatformInitialize,"   -which is a different issue than what I reported at first.  The difference may be due to me being on Xamarin.Forms 5 now.

Zahir-Khan, 2021-07-03:
I'm on xamarin 5 as well. 4.0.18 works on Android. Except for the getting stuck on read part. I forgot to mention one more thing I am now using writeasync and readasync. 

This is really nice library. I think the only one for forms that work on all three platforms. Maintaining a library like this is not easy. I wish it were though.

thhowl, 2021-07-03:
I'm using ReadAsync too.   (no writes)  I will test 4.0.18 again, on Tuesday.

thhowl, 2021-07-06:
Version 4.0.18 is now working for me with Xamarin.Forms 5.  No changes to my code since using version 4.0.8.  I cannot explain the trouble I had a few weeks back.   

@Zahir-Khan , what you are experiencing is normal / a known issue:  

Stream.ReadAsync doesn't honor cancellation token. It only checks it at the beginning. The actual operation is not guarded. As a result if remote server never responds and connection never closed it will lead to this operation hanging forever.
https://devblogs.microsoft.com/pfxteam/how-do-i-cancel-non-cancelable-async-operations/

To work around that, you have to manage your own timeout and close the connection to cancel the read.

I am closing this issue as the newer versions of the library (since 4.0.14) have resolved the issue.

closed 2021-07-06 by thhowl

thhowl, 2021-07-15:
Just noting here, on this closed issue, regarding the *different* issue that I mentioned 3 comments up: the crashing in PlatformInitialize.    It turns out, I am still experiencing the crash on my Android 9 emulator, but not on my Android 9 phone.    To say 1) this may explain my experience earlier: I may have only tried the emulator  2) there is still an issue to resolve -but it is not the same one as this issue was opened for.  I will log a separate issue AFTER I investigate my setup more, try to reproduce it on multiple emulators, multiple versions, etc.

Zahir-Khan, 2021-07-15:
@thhowl I don't think Bluetooth works on the emulator.

peterfoot, 2021-07-16:
Yes, it won't work on the emulator but it should fail gracefully.

thhowl, 2021-07-16:
@peterfoot , yes it should fail gracefully, as it does in 4.0.8.   However, version 4.0.18 crashes in construction of BlueToothClient, as I reproduced just now:

System.NullReferenceException: Object reference not set to an instance of an object.
  at InTheHand.Net.Sockets.BluetoothClient.PlatformInitialize () [0x00005] in <aa7f1e23f036426b806e750860fce0a5>:0 
  at InTheHand.Net.Sockets.BluetoothClient..ctor () [0x00006] in <aa7f1e23f036426b806e750860fce0a5>:0 
  
  But I will log as a separate issue after ensuring my emulator setup is not at fault.

peterfoot, 2021-07-17:
The code in question is an added section to enable the default radio. I think this is failing with an exception rather than returning null as expected. However, related to this, the API used to get the default adapter is deprecated in Android API 31 so it will need changing. I'm working on an update to this but haven't tested it on emulators yet