inthehand/32feet #105: Intermittent socket closing with some phones

aounmike, 2021-01-29:
I have a listener thread on my PC application using 32feet that is using BluetoothListener to accept a connection from an app on an android phone. The phone app creates a socket by calling createInsecureRfcommSocketToServiceRecord (android java). I am then able to communicate between the PC application and the phone app, sending commands back and forth. I have four phones with which to test. Unfortunately, they are all Motorola phones (moto e, moto g fast, moto g6, moto g7). On two of the phones, and only after working for long periods of time (hundreds of commands and responses sent back and forth), I receive  an exception on the PC application side during client.getstream.write or client.getstream.read that says that the socket has disconnected. It only happens on two of the four phones and it is intermittent. Furthermore, it only happens (intermittently) on those two phones if I have let them idle (let's say for around 10 seconds) without any communication between the PC and the phone. If I keep communication going (let's say one command/response each second), the problem does not occur. Just curious if anyone has seen such behavior and has a workaround (other than continually sending commands). Thanks!

polipo, 2021-01-29:
Some android version (i had found a thread but now i dont remenber it) closes the comunication if don't send anyhing. The only workaround i have found is to trasmit a blank char any second for keepalive the comunication.

aounmike, 2021-01-29:
Thank you! At least there is confirmation. I have looked and not found
anything about it. If you ever come across the thread, please let me know.

Thanks.