inthehand/32feet #108: BluetoothClient InquiryLength property

GeoffSalter, 2021-03-04:
Hello Peter

I have been using the old API version (V3.5) and I am moving to the new API. I am using the BluetoothClient class. I notice the Inquiry Length property no longer exists and the Discover Devices has been simplified. Is it possible to change the scan duration when using DiscoverDevices. It currently takes 10 seconds (which I believe is the default). 

Thanks for your help,
Regards,
Geoff

peterfoot, 2021-03-04:
Hi, I agree there needs to be a way to set this (even if only on Windows). I think the best way is to implement in the same way as v3.5 and have a property but it would likely not have any effect on other platforms.

GeoffSalter, 2021-03-04:
Hi Peter

Thanks for the quick response. Just so I understand this correctly, at the moment, with the latest version (V4) I cannot change this scan duration. Is that correct?

Regards,
Geoff

peterfoot, 2021-03-04:
Yes, sadly that is the case

GeoffSalter, 2021-03-04:
Thank you. I can work with that.

closed 2021-03-04 by GeoffSalter