inthehand/32feet #109: How do I even start using this?

YJH16120, 2021-03-17:
I've been browsing documentation for a long time, and don't even know where to start. I want to begin with getting a list of devices that are already connected.

peterfoot, 2021-03-17:
There isn't a getting started walkthrough/project and I think this is a good request to add to the list. For Bluetooth Classic (InTheHand.Net.Bluetooth) there is more code and documentation in the wild. While some of it was written for the older 3.x version most will still apply. For Bluetooth LE (InTheHand.Bluetooth) there isn't as much yet, although it follows the same model as the Web Bluetooth API and so you may find documentation for that which helps explain how everything fits together.

YJH16120, 2021-03-17:
I see, but I got pretty tired of just looking at documentation. I think having a quick start section, for Bluetooth Classic or BluetoothLE (or both) will be helpful to most.

jcs090218, 2021-03-17:
Hi, I have a small demo using this plugin in Unity C#. 

https://github.com/jcs-workspace/Test_32feet

I can work on demo project and write some document for anyone who have trouble using this plugin. :)

YJH16120, 2021-03-19:
I see. Thank you.

CHANG-CHING-CHUNG, 2021-03-22:
I understand what you feel @YJH16120 .
I searched a lot of articles online trying to get a small example about how to use 32feet.net to connect both server and client side but I got only code fragments and that didn't help...

I hope the author of this library could make a easy to understand example just like Android did.
[Here is the reference of Android Bluetooth API Overivew](https://developer.android.com/guide/topics/connectivity/bluetooth)

YJH16120, 2021-03-22:
@CHANG-CHING-CHUNG yeah, I hope he does. Currently I already switched to using the built in namespace `Android.Bluetooth` (which I didn't even know exist). And that works just fine.

jcs090218, 2021-03-22:
@CHANG-CHING-CHUNG @YJH16120 I have started the an example project with document here, https://github.com/jcs-workspace/32feet.Unity.Example. I am planning on creating more example and document for people to want to use this awesome library!

RubenGarcia, 2022-03-09:
Please add your examples to the readme.md.