inthehand/32feet #110: Getting started and Example project

jcs090218, 2021-03-20:
Hi, 32feet community!

I just started making an example project for people interested in using the library over the Unity Engine. Please checkout the repository here, https://github.com/jcs-workspace/32feet.Unity.Example.

I think this could benefit many users and especially newcomers. Since there are not many tutorials or documents out there, by having an example project and short document with an explanation could advantage people who want to get involved to this project. 

@peterfoot Would you consider accepting this repository to the org? I would want to make this project an official example for 32feet with Unity. And I would also like to work on more example projects along with tutorial document.

Let me know what you think! Thanks! :)

---

Related issue, #109.

Timeskewlol, 2024-01-18:
> Hi, 32feet community!
>
> I just started making an example project for people interested in using the library over the Unity Engine. Please checkout the repository here, [...]

HI,the url seems to be broken.I would be grateful if you could provide a new example project for unity!

jcs090218, 2024-01-18:
I've moved the repo to GitLab, see https://gitlab.com/jcs-workspace/bluetooth/32feet.Unity.Example.

Hope it helps! :)

Timeskewlol, 2024-01-18:
Thank you！I'm about to try it.Does this library support BLE？

Timeskewlol, 2024-01-18:
Thanks a lot！