inthehand/32feet #114: BluetoothClient.Connect() - Exception: The requested address is not valid in its context

JPhilC, 2021-04-19:
I have created a simple WPF app to test sending a receiving data between two PCs. It is based on the following article but with the bugs ironed out.

(https://www.c-sharpcorner.com/UploadFile/3789b7/how-to-send-data-through-bluetooth-in-a-wpf-application-usin/)

The send method is shown below:

```
public async Task<bool> Send(Device device, string content)
        {
            if (device == null)
            {
                throw new ArgumentNullException("device");
            }

            if (string.IsNullOrEmpty(content))
            {
                throw new ArgumentNullException("content");
            }

            // for not block the UI it will run in a different threat  
            var task = Task.Run(() =>
            {
                using (var bluetoothClient = new BluetoothClient())
                {
                    try
                    {
                        // connecting  
                        bluetoothClient.Connect(device.DeviceInfo.DeviceAddress, _serviceClassId);

                        // get stream for send the data  
                        var bluetoothStream = bluetoothClient.GetStream();

                        // if all is ok to send  
                        if (bluetoothClient.Connected && bluetoothStream != null)
                        {
                            // write the data in the stream  
                            var buffer = System.Text.Encoding.UTF8.GetBytes(content);
                            bluetoothStream.Write(buffer, 0, buffer.Length);
                            bluetoothStream.Flush();
                            bluetoothStream.Close();
                            return true;
                        }
                        return false;
                    }
                    catch (Exception ex)
                    {
                        System.Diagnostics.Debug.WriteLine(ex.Message);
                        // the error will be ignored and the send data will report as not sent  
                        // for understood the type of the error, handle the exception  
                    }
                }
                return false;
            });
            return await task;
        }

```
The problem I have is that the line:

`bluetoothClient.Connect(device.DeviceInfo.DeviceAddress, _serviceClassId);`

Throws an exception: The requested address is not valid in its context

Can anyone give me a pointer as to why this might be

Thanks.

JPhilC, 2021-04-19:
I have managed to get past the original error. I realised that the send and receive services were using different values for _serviceClassId.

Now though, the Connect methods returns the exception:

A socket operation was attempted to an unreachable network 4A290000000:fa87c0d0-afac-11de-8a39-0800200c9a66

Where  4A290000000 is the DeviceAddress of the receiving machine and fa87c0d0-afac-11de-8a39-0800200c9a66 is the shared _serviceClassId.

JPhilC, 2021-04-19:
Removing the receiving device and re-adding fixed the message above.

closed 2021-04-19 by JPhilC

Guangyu-Joshua-Feng, 2022-06-24:
Would you mind clarifying wht do you mean by "Removing the receiving device and re-adding fixed the message above"? I ran into the same issue as you did. See my question: https://github.com/inthehand/32feet/issues/200

JPhilC, 2022-06-24:
Hi, wow that was a while ago. I think I probably meant that I just told the receiving device (PC) to forget the bluetooth pairing and then re-paired.