inthehand/32feet #115: BluetoothClient.DiscoverDevices() hangs on Android.

JPhilC, 2021-04-20:
I have attempted to re-create the Xamarin forms BluetoothRfcomm example. But in the following code the app hangs on the line:

`foreach (BluetoothDeviceInfo bdi in client.DiscoverDevices(24))`

The complete method is shown below. The loop to show paired devices appears to work (sort of, see my later comments).

```
 protected override async void OnAppearing()
        {

            BluetoothRadio radio = BluetoothRadio.Default;
            if (radio != null && radio.Mode == RadioMode.PowerOff)
            {
                BluetoothRadio.Default.Mode = RadioMode.Connectable;
            }

            BluetoothDeviceInfo device = null;

            BluetoothClient client = new BluetoothClient();

            var devices = client.PairedDevices;

            foreach (BluetoothDeviceInfo bdi in devices)
            {
                System.Diagnostics.Debug.WriteLine(bdi.DeviceName + " " + bdi.DeviceAddress.ToString("C") + " " + bdi.Authenticated);
            }

            foreach (BluetoothDeviceInfo bdi in client.DiscoverDevices(24))
            {
                System.Diagnostics.Debug.WriteLine(bdi.DeviceName + " " + bdi.DeviceAddress.ToString("C") + " " + bdi.Authenticated);
                if (device == null)
                    device = bdi;
            }

            //client.Connect(device.DeviceAddress, BluetoothService.SerialPort);
            //stream = client.GetStream();
            //byte[] data = System.Text.Encoding.ASCII.GetBytes("Hello world!"); //! 0 200 200 210 1\r\nTEXT 4 0 30 40 Hello World\r\nFORM\r\nPRINT\r\n");
            //stream.Write(data, 0, data.Length);
            //stream.Close();

            base.OnAppearing();
        }

```

I am running the App on an Nokia 8.3 with Android 11.0 - API 30 and the aim is to connect to a Windows 10 PC. Before running the app I have added the PC as a device on the phone. Although Windows 10 says the connection was successful and the phone appears in the Bluetooth & Other Devices screen under other devices it does not appear in the list returned by Client.PairedDevices although other previous connected devices do.

Ultimately I would like to be able to pass data back and forth between an App running on the mobile and another app running on the PC but this is proving painfully difficult to achieve. I have a desktop app using 32 feet that successfully passes data between two PCs connected by Bluetooth.

I have the following permissions in the Android manifest:

```
ACCESS_BACKGROUND_LOCATION
ACCESS_COARSE_LOCATION
ACCESS_FINE_LOCATION
ACCESS_NETWORK_STATE
BLUETOOTH
BLUETOOTH_ADMIN
BLUETOOTH_PRIVILEGED

```
Any help or pointers to getting the mobile App working would be very gratefully received. I have read through the Wiki articles but found nothing there that helps.

Phil

JPhilC, 2021-04-22:
I think I found the answer in this article

[How to detect nearby Bluetooth devices with .NET and Xamarin.Android](https://jeremylindsayni.wordpress.com/2018/12/16/how-to-detect-nearby-bluetooth-devices-with-net-and-xamarin-android/)

There is a section at the bottom headed "Finally – check permissions are applied at runtime".

In the Android project I updated my MainActive.cs file with the following code:

```
            #region Get permissions at runtime (apparently needed with newer version of Android)
            const int locationPermissionsRequestCode = 1000;

            var locationPermissions = new[]
            {
                Manifest.Permission.AccessCoarseLocation,
                Manifest.Permission.AccessFineLocation
            };

            // check if the app has permission to access coarse location
            var coarseLocationPermissionGranted =
                ContextCompat.CheckSelfPermission(this, Manifest.Permission.AccessCoarseLocation);

            // check if the app has permission to access fine location
            var fineLocationPermissionGranted =
                ContextCompat.CheckSelfPermission(this, Manifest.Permission.AccessFineLocation);

            // if either is denied permission, request permission from the user
            if (coarseLocationPermissionGranted == Permission.Denied ||
                fineLocationPermissionGranted == Permission.Denied)
            {
                ActivityCompat.RequestPermissions(this, locationPermissions, locationPermissionsRequestCode);
            }
            #endregion
```

The problem then went away.

If anyone is interested I now have a basic working demonstration of communications using Bluetooth between an Android 11 device and a Windows 10 PC. The repository is [here](https://github.com/JPhilC/BluetoothXPlatformChat)

I'll leave the bug open just in case this isn't the remedy.

peterfoot, 2021-04-22:
Yes, unfortunately you need location permissions to do discovery as it is assumed that it can be used for known Bluetooth devices or beacons and reveal the location of the device. You should only need to request coarse location for this (fine location is used for GPS level location finding).
I guess an interesting question here is should the library check permissions and prompt the user when you call DiscoverDevices if the permission has not already been granted? Or should this be left entirely to the consuming application?