inthehand/32feet #118: Possible Bug - Data not being sent until the stream is closed

JPhilC, 2021-04-25:
After successfully getting a cross platform chat example working last week I have spent the best part of today trying to get a Command/Response demo working using 32Feet. However I am stuck. The Stream returned by BluetoothClient.GetStream() does not send the data when the Flush() or FlushAsync() methods are called. It only gets sent when the stream is closed. 

My original chap app, created the stream, send the text and closed the stream without waiting for a response. What I actually need is a command/response connection where the client sends a command to the server and the server returns a response which will ultimately be a chunk of JSON text.

On my client app I have the following code. My service class holds a reference to a client instance in _client.

```
        public async Task<string> Send(string command)
        {
            string response = "ERR:TIMEOUT";
            using (var stream = _client.GetStream())
            {
                StreamWriter writer = new StreamWriter(stream);
                StreamReader reader = new StreamReader(stream);
                await writer.WriteAsync(command);
                await writer.FlushAsync();
                //writer.Close();
                while (stream.DataAvailable)
                {
                    response += await reader.ReadToEndAsync();
                }
            }
            return response;
        }

```

On the server I have the following code:

```
        private async Task Listener(CancellationTokenSource token)
        {
            try
            {
                while (true)
                {
                    if (token.IsCancellationRequested)
                    {
                        return;
                    }
                    // Are we still listening for a client
                    if (_client == null)
                    {
                        _client = _listener.AcceptBluetoothClient();
                    }
                    // If we have a client see if there is new data
                    if (_client != null)
                    {
                        using (var stream = _client.GetStream())
                        using (var reader = new StreamReader(stream))
                        using (var writer = new StreamWriter(stream))
                        {
                            try
                            {
                                var command = await reader.ReadToEndAsync();
                                if (!string.IsNullOrEmpty(command))
                                {
                                    BluetoothCommandResponseEventArgs args = new BluetoothCommandResponseEventArgs()
                                    {
                                        Command = command
                                    };
                                    OnCommandReceived(args);
                                    // Now post the response
                                    await writer.WriteAsync(args.Response);
                                    await writer.FlushAsync();
                                }
                            }
                            catch (IOException)
                            {
                                break;
                            }
                        }
                    }
                }
            }
            catch (Exception ex)
            {
                // todo handle the exception  
                // for the sample it will be ignored
                Debug.WriteLine($"Exception thrown: {ex.Message}");
            }
        }

```
The problem occurs because the server code does not receive the incoming message until the client write stream is closed. If I uncomment the line `writer.Close() `the server receives the command but the client then throws an exception on the `while(stream.DataAvailable)` line because the stream is closed.

Is this a bug or am I approaching this incorrectly?

peterfoot, 2021-05-03:
I've done some testing with both a 5.1 and 8.0 device and both are showing the Flush operation working as expected. I will see if I can test on some other devices but this may be a driver specific issue rather than a particular Android version.

I think the issue may be related to the Async operations not being overridden in the AndroidNetworkStream, so FlushAsync does nothing. I'm doing some further investigation

peterfoot, 2021-05-03:
Version 4.0.17 has just been pushed to NuGet and has the missing Async methods implemented. Also it now supports the Length property to shows available data in the input stream.

JPhilC, 2021-05-09:
Great. I'll give it a try. Thanks.

closed 2021-05-09 by JPhilC