inthehand/32feet #119: Xamarin Android - NetworkStream.WriteAsync does not work.

uksreejith, 2021-05-02:
Hi.

The below code works normally in Windows but does not work in Xamarin Android.

```
NetworkStream networkStream = _btClient.GetStream();
var bytesToWrite = Encoding.ASCII.GetBytes(data);
await networkStream.WriteAsync(bytesToWrite, 0, bytesToWrite.Length).ConfigureAwait(false);
```

However in Xamarin.Android the following code works:

```
StreamWriter sw = new StreamWriter(networkStream, System.Text.Encoding.ASCII);
sw.WriteLine(data);
sw.Flush();
sw.Close();
```

Any insight on why this is so?

With Regards,
Sreejith Krishnadas

peterfoot, 2021-05-02:
What version of Android are you running on? I've read that Flush does nothing on Android 4.x and older.

peterfoot, 2021-05-02:
Possibly a duplicate of #118 - the Flush operation on the stream is not behaving as expected

uksreejith, 2021-05-03:
I have been testing on an Android 6.0 device.

peterfoot, 2021-05-03:
Version 4.0.17 has just been pushed to NuGet and has the missing Async methods implemented. Also it now supports the Length property to shows available data in the input stream.

peterfoot, 2021-05-06:
Going to close this now, if you encounter the issue in 4.0.17 please feel free to reopen

closed 2021-05-06 by peterfoot

uksreejith, 2021-05-09:
Hi @peterfoot. Sorry for the late response. Was a bit busy. Just checked with the latest Nuget update and its working fine now. 👍