inthehand/32feet #125: Connecting PC to Android Galaxy S20 via Bluetooth

royben, 2021-05-17:
I am trying to connect and send data from my PC to my android phone.

Service side (Xamarin Forms)

`listener.AcceptBluetoothClient();` is not blocking and always returns null.

```csharp
BluetoothListener listener = new BluetoothListener(BluetoothService.SerialPort);
listener.Start();

BluetoothClient client = null;

while (client == null)
{
    Thread.Sleep(10);
    client = listener.AcceptBluetoothClient();
}

var stream = client.GetStream();

do
{
    byte[] data = new byte[1];
    stream.Read(data, 0, data.Length);
    Debug.WriteLine(Encoding.UTF8.GetString(data));
} while (stream.DataAvailable);

listener.Dispose();
```

PC side (Console App)

The `Connect` method throws exception:
`System.Net.Sockets.SocketException: 'The requested address is not valid in its context FCDE90B69C4C:00001101-0000-1000-8000-00805f9b34fb (0)'
`

```csharp
var client = new BluetoothClient();

BluetoothDeviceInfo info = null;

foreach (var device in client.PairedDevices)
{
    if (device.DeviceName.Contains("Galaxy"))
    {
        info = device;
        break;
    }
}

client.Connect(info.DeviceAddress, BluetoothService.SerialPort);
var stream = client.GetStream();

byte[] data = Encoding.UTF8.GetBytes("Hello World");
stream.Write(data, 0, data.Length);

client.Dispose();

```

If I run the service side on a PC, the `AcceptBluetoothClient();` **will block** as expected.

Any advice would be appreciated. Thanks!

royben, 2021-05-17:
Now I see...

```csharp
// 32feet.NET - Personal Area Networking for .NET
//
// InTheHand.Net.Sockets.BluetoothListener (Android)
// 
// Copyright (c) 2018-2020 In The Hand Ltd, All rights reserved.
// This source code is licensed under the MIT License

using Android.Bluetooth;
using InTheHand.Net.Bluetooth;
using System;

namespace InTheHand.Net.Sockets
{
    partial class BluetoothListener
    {
        void DoStart()
        {

        }

        void DoStop()
        {

        }

        bool DoPending()
        {
            return false;
        }

        BluetoothClient DoAcceptBluetoothClient()
        {
            return null;
        }
    }
}
```

The home page states that the BluetoothListener is supported on Android...

royben, 2021-05-17:
Here is a very rough implementation..

```csharp
partial class BluetoothListener
{
    private BluetoothAdapter _adapter;
    private BluetoothServerSocket _server;

    void DoStart()
    {
        _adapter = BluetoothAdapter.DefaultAdapter;

        _server = _adapter.ListenUsingRfcommWithServiceRecord(ServiceName, UUID.FromString(serviceUuid.ToString()));
    }

    void DoStop()
    {
        _server?.Dispose();
        _adapter?.Dispose();
    }

    bool DoPending()
    {
        return false;
    }

    BluetoothClient DoAcceptBluetoothClient()
    {
        return new BluetoothClient(_server.Accept());
    }
}
```

peterfoot, 2021-05-17:
Thanks, I'm looking into adding this shortly.

closed 2023-10-11 by peterfoot