inthehand/32feet #128: Operation is not supported in this platform

orcunor, 2021-05-27:
[image]

I installed InTheHandBluetoothLe and microsoft.windows.cswinrt libraries and only wrote a single method and got this error.

stephen-systemfriend, 2021-06-14:
You'll need to give more information - is this a donet standard or a dotnet core project, are you using unity (and if so, mono etc) and maybe some other things.

David-1251, 2021-08-18:
I get this same exception when using this method in .NET 5.

r-pankevicius, 2022-03-21:
Should be PlatformNotSupportedException due to https://docs.microsoft.com/en-us/dotnet/core/compatibility/interop/5.0/built-in-support-for-winrt-removed

peterfoot, 2022-03-22:
What is your target platform - which Windows 10 version? Is it older than 10.0.19041?

closed 2025-12-14 by peterfoot