inthehand/32feet #132: [QUESTION] Connect / Disconnect

steam3d, 2021-06-11:
[image]
Can i connect / disconnect programmatically headphones?
I found the way to connect using https://docs.microsoft.com/en-us/windows/win32/api/bluetoothapis/nf-bluetoothapis-bluetoothsetservicestate, but it takes a lot of time and device can't be disconnected even if i disable all services.

peterfoot, 2021-06-14:
Historically BluetoothSetServiceState was used to enable/disable device manager entries for specific services e.g. virtual serial ports. I wouldn't recommend using it to toggle a headset. I will look to see if there is an alternative API which can be used to force a connection. It feels like this should be controlled via an Audio API by selecting a different default device but I'm not sure what APIs are available for this.

steam3d, 2021-06-14:
I already tried Audio APIs, there is no way to change state of mmdevice. 

There is a partial way to connect (but you can't disconnect device)
```
 using Windows.Devices.Bluetooth;
 async void ConnectBTUWP()
 {
   btdevice = await BluetoothDevice.FromIdAsync(BTId);
   rfcommResult = await btdevice.GetRfcommServicesAsync();
   rfcomm = rfcommResult.Services[0];
    
     if (btdevice.ConnectionStatus != BluetoothConnectionStatus.Connected)
       {
         // Create a socket and connect to the target
         var _socket = new StreamSocket();
         await _socket.ConnectAsync(
         rfcomm.ConnectionHostName,
         rfcomm.ConnectionServiceName,
         SocketProtectionLevel.BluetoothEncryptionAllowNullAuthentication);
       }    
 }
```

The only one dirty way to connect or disconnect device is disable all services or enable all services of device.

MarshalT, 2021-12-22:
I'm having the same problem, I want to disconnect my computer from the Bluetooth connection or program it to disable it

PolarGoose, 2022-01-24:
I'm struggling with finding a solution for this problem as well.
I have created a https://github.com/PolarGoose/BluetoothDevicePairing console utility where I want to implement connection to devices the same way it is done via Window GUI.
I have asked the same question on StackOverflow ([How to connect to a paired audio Bluetooth device using Windows UWP API?](https://stackoverflow.com/questions/62502414/how-to-connect-to-a-paired-audio-bluetooth-device-using-windows-uwp-api)) but haven't got any answer.

### Doesn't work: using [BluetoothSetServiceState](https://docs.microsoft.com/en-us/windows/win32/api/bluetoothapis/nf-bluetoothapis-bluetoothsetservicestate).
The fact that it doesn't work you can test yourself without any code:
* Connect to your device using Windows GUI.
* Open `Control Panel\Hardware and Sound\Devices and Printers`.
* Select your Bluetooth device there and right click -> Properties.
* Open "Services" tab as it is show on the screenshot:
[image]
You will see all services which your device provides.
* Uncheck all checkboxes and click "Apply".
* The device will be disconnected. But it breaks Windows's pairing GUI
[image]
* Therefore it is not what Windows does when you click "Disconnect" button. They do something else, because if you disconnect from the device via Windows pairing dialog, the Bluetooth services of the device will not be disabled.
* But what is worse is that after you check all checkboxes in the "Service" tab back, it doesn't often make Windows connect to the device properly. I observed that the Windows pairing dialog will show the status of a device "Connected" but not "Connected as voice, music" and no sound device will be present for the headphones.

### Doesn't work: If device is paired but not connected, unpair it first and then immediately pair again
In `Windows.Devices.Enumeration` namespace there are two methods [PairAsync](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.deviceinformationcustompairing.pairasync?view=winrt-22000) and [UnpairAsync](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.deviceinformationpairing.unpairasync?view=winrt-22000). If device hasn't been paired before and you call `PairAsync` Windows will pair and connect to the device. But if device is paired but not connected, `PairAsync` will just fail.
But if we call `UnpairAsync` in this case and then after it finishes call `PairAsync`, Windows will pair and connect to the device. But it doesn't work reliably. In my case it only connected to my device as a "Music" but without "Voice".

peterfoot, 2022-01-24:
There isn't currently an API (Win32 or WinRT/UWP) to do this. As already stated BluetoothSetServiceState is not appropriate for this and should not be used.
If the reason for disabling the headset is to play audio through a different endpoint you can do this by setting the AudioDevice of a MediaPlayer object. Use the DeviceInformation.FindAllAsync() method with a Windows.Media.Devices.MediaDevice.GetAudioRenderSelector() selector to enumerate all output devices, and then choose the one you want which isn't the headset. However apps can't disconnect a headset device.

PolarGoose, 2022-01-24:
@peterfoot,
Thank you for your reply. Do you know what happens when a user clicks "Connect" button the Windows GUI? What mechanism Windows uses?

peterfoot, 2022-01-24:
Windows connects via either Headset, Handsfree or A2DP profiles depending on the device. This causes a separate SCO channel to be opened for the audio (the main profile being an RFComm channel for control only). Trying to open a socket on one of those profiles yourself will not allow the normal Windows support for these to work, including the separate audio channel.

steam3d, 2022-01-29:
Has anyone tried directly requesting information from Microsoft?

PolarGoose, 2022-01-31:
@steam3d,
I haven't tried to do that. Do you know how to do that?

peterfoot, 2022-01-31:
The current place to ask technical questions is Microsoft Q&A - https://docs.microsoft.com/en-us/answers/topics/windows-api.html

PolarGoose, 2022-02-01:
@peterfoot ,
Thank you for the link, I have asked created a topic: https://docs.microsoft.com/en-us/answers/questions/718643/connect-to-a-paired-bluetooth-device-from-c.html

PolarGoose, 2022-03-19:
I think I have found something that works, even though it is still a workaround:
Headphones have PnP devices related to them. Toggling states of these devices will make Windows connect to the headphones.
That is the powershell script which I tested.
Preconditions: Headphones should be paired but not connected.
```
$headphonesName = "WH-1000XM3"

$bluetoothDevices = Get-PnpDevice -class Bluetooth
$headphonePnpDevices = $bluetoothDevices | Where-Object { $_.Name.StartsWith("$headphonesName") }

if(!$headphonePnpDevices) {
    Write-Host "Coudn't find any devices related to the '$headphonesName'"
    Write-Host "Whole list of available devices is:"
    $bluetoothDevices
    return
}

Write-Host "The following PnP devices related to the '$headphonesName' headphones found:"
# ForEach($d in $headphonePnpDevices) { Write-Host $d.Name " | " $d.DeviceID }
ForEach($d in $headphonePnpDevices) { $d }

Write-Host "`nDisable all these devices"
ForEach($d in $headphonePnpDevices) { Disable-PnpDevice -InstanceId $d.InstanceId -Confirm:$false }

Write-Host "Enable all these devices"
ForEach($d in $headphonePnpDevices) { Enable-PnpDevice -InstanceId $d.InstanceId -Confirm:$false }

Write-Host "The headphones should be connected now."
Write-Host "It may take around 10 seconds untill the Windows starts showing audio devices related to these headphones."
```

PolarGoose, 2023-08-12:
@shunf4,
The link https://github.com/shunf4/ToothTray doesn't work

shunf4, 2023-08-12:
> The link https://github.com/shunf4/ToothTray doesn't work.

Sorry, it's https://github.com/shunf4/ToothTrayCli .

shunf4, 2023-08-12:
> I have also briefly tested the original [m2jean/ToothTray](https://github.com/m2jean/ToothTray). Unfortunately, it doesn't work correctly. When I disconnect my Headset it still has a status **Connected** in the **Windows Bluetooth & Devices dialog**

Can you try if it works in Windows 10?

BTW seems like Sony's headsets really have quirks with them; IIRC I tried using one in Ubuntu Linux but failed;  the headset just won't connect.

PolarGoose, 2023-08-13:
@shunf4,
> BTW seems like Sony's headsets really have quirks with them

It is not related Sony, the same problem happens with **Bose 700** and **Jabra Evolve 75e**
[image]
[image]

Unfortunately, I don't have Windows 10 to test if **ToothTray** works there.

seanyu0, 2024-01-20:
> @shunf4, The link https://github.com/shunf4/ToothTray doesn't work.
> 
> I have also briefly tested the original [m2jean/ToothTray](https://github.com/m2jean/ToothTray). Unfortunately, it doesn't work correctly. When I disconnect my Headset it still has a status **Connected** in the **Windows Bluetooth & Devices dialog** ![image](https://private-user-images.githubusercontent.com/35307286/260206230-dfc41996-26a8-414a-8014-36c4ef8de677.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU3ODUzMDEsIm5iZiI6MTcwNTc4NTAwMSwicGF0aCI6Ii8zNTMwNzI4Ni8yNjAyMDYyMzAtZGZjNDE5OTYtMjZhOC00MTRhLTgwMTQtMzZjNGVmOGRlNjc3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTIwVDIxMTAwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRjNmJiM2VjNjNmYjE3ZTRlNjU4ZDRiZjc5NWI0ZDNiNzc2YmZjNjE2YzViMWQyYjYxMmNjYjhiMTZkZTE4MzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ykW0UcfkM8Opt6hJI_PezcMocky2duwmLL8sDIbhTtk) Or it disconnects only audio but leaves the microphone connected: ![image](https://private-user-images.githubusercontent.com/35307286/260206379-2771f78d-4d2a-4314-8eb1-4b6d2093f745.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU3ODUzMDEsIm5iZiI6MTcwNTc4NTAwMSwicGF0aCI6Ii8zNTMwNzI4Ni8yNjAyMDYzNzktMjc3MWY3OGQtNGQyYS00MzE0LThlYjEtNGI2ZDIwOTNmNzQ1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTIwVDIxMTAwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE0ZDQ5NThkZWMyMmQwZWM2ZGQ2ZGI0MzA0MzE2NzllN2FiOWQ5NjAzMTdmMWZhZGYzN2QzMjQ2NDkyMDI3ZjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qYopeqvzvesj96d1gpwE_aJz95rbI9RAE1Z2xxzw-y0)

https://github.com/m2jean/ToothTray is the correct way to connect and disconnect bluetooth audio device. But in win11, this way can only enumerate headset service, lost headsfree service, so connect or disconnect doest not work properly. Just change IMMDeviceEnumerator::EnumAudioEndpoints parameter EDataFlow from eCapture to eAll, this problem will be fixed.

PolarGoose, 2024-01-21:
> @shunf4, The link https://github.com/shunf4/ToothTray doesn't work.
>
> I have also briefly tested the original [m2jean/ToothTray](https://github.com/m2jean/ToothTray). Unfortunately, it doesn't work [...]

I have created a bug for this issue: https://github.com/m2jean/ToothTray/issues/10