inthehand/32feet #134: BluetoothLE find devices crashes with an ArgumentException

stephen-systemfriend, 2021-06-29:
If you run the below code (Console app with the Nuget package) on windows 10.0.19041 , it will crash with the following error. 
code
``` C#
using System;
using System.Threading.Tasks;
using InTheHand.Bluetooth;

namespace InTheHandBLETest
{
    class Program
    {
        static void Main(string[] args)
        {
            var discoveryTask = TestDeviceDiscovery();
            discoveryTask.Wait();
        }

        private static async Task<bool> TestDeviceDiscovery()
        {
            var discoveredDevices = await Bluetooth.ScanForDevicesAsync(new RequestDeviceOptions
            {
                AcceptAllDevices = true
            });
            Console.WriteLine($"found {discoveredDevices?.Count} devices");
            return discoveredDevices?.Count > 0;
        }
    }
}

```
error
```
System.AggregateException
  HResult=0x80131500
  Message=One or more errors occurred. (The parameter is incorrect.

The provided device ID is not a valid BluetoothLEDevice object.)
  Source=System.Private.CoreLib
  StackTrace:
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.Tasks.Task.Wait()
   at InTheHandBLETest.Program.Main(String[] args) in C:\Users\stephen\source\repos\InTheHandBLETest\Program.cs:line 12

  This exception was originally thrown at this call stack:
    [External Code]
    InTheHandBLETest.Program.TestDeviceDiscovery() in Program.cs

Inner Exception 1:
ArgumentException: The parameter is incorrect.

The provided device ID is not a valid BluetoothLEDevice object.
```

the same code worked last christmas, so I think an update killed it. Any chance of fixing this?

stephen-systemfriend, 2021-06-29:
I ran the above code directly in the repo code so I could debug it better, and it seems that one device that my PC thinks is connected (even though it isn't, and it reports an error when I attempt to click remove device) is causing the error. I guess this is primarily a problem with my environment, though it would be amazing if the code could catch that exception and keep going, rather than blowing up and taking the app down with it.

stephen-systemfriend, 2021-06-30:
I produced a fix that works in my enviroment in #135 . The root cause of the problem seems to be the broken driver or whatever that has left me with the zombie device, but with this fix at least this can still run even in the presence of that mess.

peterfoot, 2021-06-30:
Thanks for the pull request, I'll go ahead and close the issue now.

closed 2021-06-30 by peterfoot

bbhxwl, 2021-07-21:
How did you solve it? I had the same problem, I couldn't search the device.

stephen-systemfriend, 2021-07-21:
@bbhxwl there is a fix in the above pull request. I don't know when the version containing that PR will be released, but until then you can built it yourself. 
Alternatively, if you are able to, disconnect all the bluetooth devices and make sure your PC has forgotten them.

bbhxwl, 2021-07-21:
> @bbhxwl there is a fix in the above pull request. I don't know when the version containing that PR will be released, but until then you can built it yourself.
> Alternatively, if you are able to, disconnect all the bluetooth devices and make sure your PC has forgotten them.

Can I download the latest project and compile it myself? There is a previously paired one in my Bluetooth device, which can't be deleted. I don't know why.

stephen-systemfriend, 2021-07-21:
yep, just clone the branch at https://github.com/stephen-systemfriend/32feet/tree/invalid-device-id-check.
I'm currently actually using a project reference to that from visual studio while I wait for the next release

peterfoot, 2021-07-21:
Sorry for the delay, the latest version 4.0.20 should be appearing on NuGet now.

bbhxwl, 2021-07-21:
> 是的，只需在https://github.com/stephen-systemfriend/32feet/tree/invalid-device-id-check克隆分支。
> 在等待下一个版本时，我目前实际上正在使用来自 Visual Studio 的项目引用

Is there a compiled DLL for WinForm? I can't compile here. Very urgent, if you have compiled, can you send me one?

seriousness code explain project file that 's ok Disable display status

error XA5207 Android jar with API level 27 not found. This means that the Android SDK platform with API level 27 is not installed. Please install it in Android SDK manager Tools > Android > Android SDK Manager... Or change the xamarin.android project to target the installed API version（ Missing C: (program files (x86)  Android  Android SDK  platforms  android-27  Android. Jar.) InTheHand.BluetoothLE C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Xamarin\Android\Xamarin.Android.Common.targets five hundred and six

peterfoot, 2021-07-21:
The version with the fix is now on NuGet - https://www.nuget.org/packages/InTheHand.BluetoothLE/

bbhxwl, 2021-07-21:
> The version with the fix is now on NuGet - https://www.nuget.org/packages/InTheHand.BluetoothLE/

Thank you very much, my English is not good, China can not search WinForm related Bluetooth program, thank you for your efforts.

bbhxwl, 2021-07-21:
> The version with the fix is now on NuGet - https://www.nuget.org/packages/InTheHand.BluetoothLE/
ScanForDevicesAsync and  RequestLEScanAsync

What's the difference?
What is the event of asynchronously scanning ble?

peterfoot, 2021-07-21:
RequestLEScan is a WebBluetooth method which returns a BluetoothLEScan which performs continuous scanning until stopped and raises events in the main Bluetooth object. ScanForDevicesAsync does a one-off scan of available devices and returns a collection. It's provided by the library for convenience but has no equivalent in the Web Bluetooth standard which the library tries to follow.