inthehand/32feet #138: Why is ble connection failed?

bbhxwl, 2021-07-21:
Ble has not been paired. After searching Bluetooth, direct connection fails? I developed it for the first time

[image]

peterfoot, 2021-07-21:
Please try after manually pairing the device, it may be that the device must be paired before accepting connections (which is common for a lot of device types)

bbhxwl, 2021-07-21:
> Please try after manually pairing the device, it may be that the device must be paired before accepting connections (which is common for a lot of device types)

In win10 system settings, when Bluetooth is added manually, you will be prompted [please try to reconnect the device]. I can connect to this ble using IOS ble debugging tool.

Can I search for paired bluetooth by using only inthehand.bluetooth? I didn't find it. If I have connected to a device once, what method do I use to connect to the device directly without searching next time? I am a novice, English is not good, using translation software. Please help me.

peterfoot, 2021-07-21:
Bluetooth.GetPairedDevicesAsync will return a collection of paired devices and because there is no discovery it is much faster.

bbhxwl, 2021-07-21:
If you use 【ScanForDevicesAsync】 to get the scan list, and then use 【Gatt.ConnectAsync()】to connect, it will not appear in 【GetPairedDevicesAsync 】, right?   If you know the device ID, can you connect directly?

closed 2021-07-24 by bbhxwl